tio/tio #18: Issues with the quit sequence

jwilk, 2016-05-07:
There are two related issues with the quit sequence (*ctrl+t q*):

* There is no way to quit tio without sending *ctrl+t* to the device, which might have unwanted side effects. (And if *ctrl-g* does something stupid, you won't even see it, because you've just quit.)

* There is no way to send the *ctrl+t q* sequence to the device.

lundmar, 2016-05-08:
First point is fixed by 755ac53553826b0588e1cb90f63ffca282916460

I consider ctrl-t q fairly unique so I don't think we need to support sending it. That being said, it if ever becomes an issue I will fix it :)

jwilk, 2016-05-08:
755ac5355382 doesn't seem to fix it for me.
I still see ^t being sent to the device in strace output:
```
_newselect(4, [0 3], NULL, NULL, NULL)  = 1 (in [0])
read(0, "\24", 1)                       = 1
write(3, "\24", 1)                      = 1
_newselect(4, [0 3], NULL, NULL, NULL)  = 1 (in [0])
read(0, "q", 1)                         = 1
```

And yeah, I don't have practical need for sending *ctrl-t q*, but IMO it should be implemented for completeness. :)

lundmar, 2016-05-08:
Small indexing error, my bad. Fixed in 7c493d957c65c831790fc0051528525ee6cbc51b

closed 2016-05-08 by lundmar