tio/tio #27: Poor error handling for stdin reads

jwilk, 2016-05-09:
If read() from stdin fails, then tio prints a warning:
```
status = read(STDIN_FILENO, &c_stdin[0], 1);
if (status < 0)
    printf("Warning: Could not read from stdin\r\n");
```
...but then it proceeds as if nothing bad happened:
```
/* Exit upon ctrl-t + q sequence */
c_stdin[2] = c_stdin[1];
c_stdin[1] = c_stdin[0];
```
But if an error occurred, `c_stdin[0]` is either uninitialized or initialized by a previous successful read. :worried:
Also, if `read()` failed, then next `read()` is very likely to fail again, and again... flooding screen with warnings.
It would be better to just stop the program if a stdin read error occurred.

lundmar, 2016-05-09:
True. The current strategy is a bit too optimistic hoping that maybe the subsequent read would succeed. If stdin breaks it is likely broken for the rest of the session so exit in this case is probably the best option.

lundmar, 2016-05-09:
Upgraded warning to error in ad551b363eb5aa1f464488cd189f655b5587625d

closed 2016-05-09 by lundmar