tio/tio #42: tty.c:221: error: 'B500000' undeclared

jwilk, 2016-05-25:
tio fails to build on FreeBSD:
```
gcc -DHAVE_CONFIG_H -I. -I../src/include     -Wall -g -O2 -MT tty.o -MD -MP -MF .deps/tty.Tpo -c -o tty.o tty.c
tty.c: In function 'tty_configure':
tty.c:221: error: 'B500000' undeclared (first use in this function)
tty.c:221: error: (Each undeclared identifier is reported only once
tty.c:221: error: for each function it appears in.)
tty.c:224: error: 'B576000' undeclared (first use in this function)
tty.c:230: error: 'B1000000' undeclared (first use in this function)
tty.c:233: error: 'B1152000' undeclared (first use in this function)
tty.c:236: error: 'B1500000' undeclared (first use in this function)
tty.c:239: error: 'B2000000' undeclared (first use in this function)
tty.c:242: error: 'B2500000' undeclared (first use in this function)
tty.c:245: error: 'B3000000' undeclared (first use in this function)
tty.c:248: error: 'B3500000' undeclared (first use in this function)
tty.c:251: error: 'B4000000' undeclared (first use in this function)
```

POSIX defines only constants up to `B38400`, so anything above that should be probably guarded by an ifdef.

lundmar, 2016-05-25:
I'm considering creating some configure tests to probe which speeds are available and handle the code accordingly. Ultimately, the available speeds also need to be reflected in the completion script.

lundmar, 2016-05-26:
Ok, so I've cooked up a generic baud rate detection mechanism to avoid introducing platform specific handling. It should solve this issue. Please see 8eb0e6d87457071c5b02643b9b3101959c24d32f

lundmar, 2016-05-26:
Follow up fix 933792d73f6fabb18c6b1b377cbf60de25b8f307

jwilk, 2016-05-26:
933792d73f6f builds fine on FreeBSD.

lundmar, 2016-05-26:
Great - closing issue.

closed 2016-05-26 by lundmar