tio/tio #45: Single place for list of baud rates

jwilk, 2016-05-26:
In 0f02e6e551cb, I reduced the number of places that has a list of baud rates from 4 to 2. But IMO that's still one too many. Having the list in multiple places is error-prone.
It'd be cool if this highly repetitive switch body in `tty_configure()` could be generated automatically by the configure script.

lundmar, 2016-05-27:
While I agree having a single source of baud rate configuration is a sound idea I do not agree it is such a good idea to start auto generating C code for this simple case.

I want to confine all tty related stuff in tty.c and I don't think the added complexity and disruption of introducing eg. a tty.c.in template to support generating tty.c is worth it. For the sake of code readability and locality, when looking exclusively at tty.c I think it is actually a good thing that is clear what options are supported in tty_configure() - with auto generated C code that would get lost for the baud rate section.

Now, is the current switch body pretty? Maybe not, but it is hard to make it less repetitive because of the nature of what is going - conversion of supported baud rates. If we had code introspection support like python does we could make it a one liner but C does not. At best, without generating C code, we could maybe introduce a macro to shorten the switch body lines even more but even that I think might be one abstraction too far which will just make the switch body look odd.

jwilk, 2016-05-27:
Fair enough.
How about making the configure script extract the list of baud rates from `src/tty.c`?

lundmar, 2016-05-27:
Hmm, maybe if there was a pretty way to do that but it would kind of turn the classic configure model upside down. Configure providing configuration to the code base is fine but the other way seems wrong to me.

To be honest, I'm okay with maintaining the two lists aligned - one for configure detection, one for implementation. I'm just not a fan of going out of the way to generate tty.c sorry :smile:

lundmar, 2016-05-27:
One possible alternative solution (that does not introduce tty.c.in) would be to AC_DEFINE some elaborate macro which expands into the various cases in the switch body.  However, I'm not sure that kind of m4 magic is even possible nor pretty and we would still lose the baud rate overview in tty_configure().

lundmar, 2016-05-28:
Ok, I'm going ahead with my suggested compromise solution to obtain a single source of baud rate configuration and close this issue. It avoids the tty.c.in construction.

I've put it in a separate branch - please see 08191890f9ff108205e92824cef30a5919eb9701

Also, it enables detection of all baud rates - just in case we are wrong in our assumption of all baud rates below 57600 being supported.

Please review the solution - if okay I will merge it in. Thanks.

jwilk, 2016-05-28:
I'd explicitly initialize `BAUDRATE_CASES` to empty string, so that setting this variable in evironment doesn't break anything.
In not very fond of increased number of configure probes, because they are slooow. But oh well, it's not a big deal.
Other than that, looks good to me! :+1:

lundmar, 2016-05-28:
Done.

Well, yes - the configure time is increased slightly but it is still fast, even on those highly loaded Travis CI build servers :smile:

closed 2016-05-28 by lundmar