tio/tio #49: non-standard baud rates don't work

jwilk, 2016-06-10:
`tty_configure()` looks like this (with irrelevant parts snipped):

```c
    speed_t baudrate = 0;
    memset(&tio, 0, sizeof(tio));
    switch (option.baudrate)
    {
        AUTOCONF_BAUDRATE_CASES
        default:
            standard_baudrate = false;
            break;
    }
    if (standard_baudrate)
    { /* ... */ }
    } else
        tio.c_ispeed = tio.c_ospeed = baudrate;
        tio.c_cflag &= ~CBAUD;
        tio.c_cflag |= BOTHER;
    }
```
But if the baud rate is non-standard, then `baudrate` remains set 0, and then `tio.c_ispeed` and `tio.c_ospeed` become 0 too.
The assignment was probably meant to be:
```c
        tio.c_ispeed = tio.c_ospeed = option.baudrate;
```
However, I don't believe the above change would help much.
In [Debian bug #683826](https://bugs.debian.org/683826), Ian Jackson wrote:
> This does not currently work properly with glibc on Linux even if you do it by hand because glibc uses the Linux ioctls `TGGETS`/`TCSETS` whose Linux termios struct does not contain `c_ispeed` and `c_ospeed`.

lundmar, 2016-06-10:
Right, it should be option.baudrate.

For the arbitrary baud rate support I'll change the code so that, in case of linux, the correct ioctl is directly used to configure speed as Jackson describes.

Thanks for pointing this out - I didn't read Jacksons post carefully first time so I didn't catch it.

lundmar, 2016-06-10:
Ok, after investigating this feature further I've decided to abandon support for arbitrary baud rates because glibc does not provide any means for it and using the bare metal kernel interface will not work on older Linux kernel versions. Even with recent Linux kernels it will only work in combination with very few drivers. So, better safe than sorry and go with portable code. This decision can of course be revisited if anyone really really wants this feature.

Feature reverted by 61655faaa8a491d9f9dfac2e5530c930ed2668f3

jwilk, 2016-06-10:
I agree that supporting arbitrary baud rates is not worth the effort.
Feel free to close the issue.

closed 2016-06-10 by lundmar