tio/tio #52: Zmodem support

bestouff, 2016-08-22:
Hi !

Would it be possible to add X|Y|Zmodem support to tio ?

Regards,

        Xav

lundmar, 2016-08-22:
Hi Xav,

I'm sorry but it is not a planned feature. Tio is originally meant to be very simple and only support basic tty input and output - meaning no support for advanced modem features such as X/Y/Zmodem file transfers. For such features I would look to eg. minicom, picocom, rz/sz etc..

However, I'm a bit undecided on this. If enough people think that tio should support such a feature and if it can be added in an elegant and simple way I might reconsider.

Either way, I have no plans to (re)implement the X/Y/Z protocols but sending and receiving files could be added as key commands which calls external tools such as rz/sz etc. directly. Unfortunately, since they are external tools, I suspect their output will not match nor integrate nicely with the output style of tio.

Br, Martin

closed 2016-08-26 by lundmar

lmunch, 2020-05-14:
+1 for some kind of file transfer support

dustymabe, 2025-04-29:
Note Xmodem and Ymodem were eventually implemented (search for `modem` in the docs).

kgraefe, 2025-05-05:
that's great :-) the corresponding PR is https://github.com/tio/tio/pull/208