tio/tio #54: no "configure" on raspberry pi

joostmarkerink, 2017-02-20:
"./configure" answers with "-bash: ./configure: No such file or directory"
running debian jessy on raspberry pi 2

larseje, 2017-02-21:
Sounds like you may have downloaded a git archive instead of a release. If you download from this page https://github.com/tio/tio/releases you need to download using the labels named something like "tio-1.20.tar.xz" instead of the label "Source code". The latter will give you an archive of the files tracked by git, and the configure script is not included in git but rather created as part of the release.

If you have acquired the sources from somewhere else, could you please share that location?

joostmarkerink, 2017-02-21:
No worries, I just downloaded your git source as zip. 

And in it there is the same install instructions.
Thanks for the information, now I installed it perfectly from the “release" source code. Maybe to avoid confusion you can update the install instructions or add the configuration command to the source code.
It works great. I use it to communicate from the raspberry pi to an arduino.
keep up the good work,

Joost

closed 2017-03-26 by lundmar