tio/tio #62: Send \r\n instead of just \n?

larsks, 2017-11-02:
I am communicating with a serial device that wants newlines to be sent as `\r\n` rather than just `\n`. Is there any way to get `tio` to perform that translation? Without that I can't actually execute any commands on the device to which I am connected.

lundmar, 2017-11-02:
I'm sorry but tio does not support remapping of special characters. For such feature I recommend that you look at tools like picocom which features all sorts of advanced options including input/output character remapping.

That being said, I might make it possible for users to indirectly configure character remapping using e.g. 'stty -F <serial device> -onlcr' before starting tio on same serial device - that should in principle allow for what you are asking. Unfortunately, for now, tio will override such settings.

Last option would be to figure out a way to reconfigure the tty settings of your target device.

sur5r, 2017-11-02:
Given tio is a minimalistic tool it might not be a good idea to start implementing too many features.

That said, there are some ideas how this could be accomplished:

1. `tio` could grow flags to to all sorts of CR/LF translation.
2. `tio` could grow a flag to skip `tcsetattr` and keep whatever was set before. This can only work if the device is already present when starting tio.
3. `tty_configure()` could start off the old `struct termios` and only modify the speed/bits/handshake settings. This also requires the device to be present.

The first option might be simple, but at some point tio might end up reimplementing most of stty. I doubt that's a good idea.

Thes second option is most probably the smallest possible code change but might confuse users because speed/etc. flags will suddenly have no effect.

The third option, also to be triggered by a new option, would be the most elegant way.

lundmar, 2017-11-02:
Right. I feel that any character remapping feature should be complete and support all sorts of mappings in which case it will become quite an extensive and complex interface and that would go against the simplistic nature of tio. I really don't want to go down that road. Also, I think it is a feature that only few people would use.

However, I think we should support the possibility of manually configuring the tty device using stty in case it is needed for special situations like this. Meaning, option 2. or 3. would be a possibility. I'm thinking option 3. would be the best because it would not require any new options but I'm not sure if it might cause trouble for others. Ultimately we could add a --skip-init to support option 2, in which case speed/databit etc. settings will have no effect.

larsks, 2017-11-02:
For what it's worth, just about every serial device I've encountered expect `\r\n` instead of just `\n`.  Adding flags for common `tcsetattr` flags doesn't seem like feature creep to me.

lundmar, 2017-11-02:
@larsks I'm curious what type of device are you connecting there?

Also, are the strings your receive terminated with `\r\n` ?

All of my Linux targets work with tio in its current configuration. You are the first to report this issue but if it is more common than I perceive then I might add a single option to support this specific mapping.

I've done a little testing and found out that option 3 is no good - it messes up everything for your typical Linux target. I'm now considering a --skip-init or --crnl option.

lundmar, 2017-11-04:
I went through the doc for all the configurations flags for tty terminals and found the ones related to remapping/translation of characters:
`INLCR    Translate NL to CR on input.`
`IGNCR   Ignore carriage return on input.`
`ICRNL   Translate carriage return to newline on input (unless IGNCR is set).`
`ONLCR   Map NL to CR-NL on output.`
`OCRNL   Map CR to NL on output.`

What we can do is add a single option to expose these configuration flags for input or output remapping. In this particular case I expect something like this would do the job:

`--remap=onlcr,igncr`

Assuming that `\r\n` needs to be sent and that we receive `\r\n` in which case `\r` needs to be ignored. 

I realize that there are a lot of non-linux targets out there and with this fairly simple option we can have tio support them all with whatever combination they need to handle CR/NL.

Let me know what you think.

lundmar, 2017-11-05:
I've implemented and added the special character mapping feature.

Specifically, I've added a single --map option through which both input and output mappings can be defined:

       -m, --map <flags>

              Map (replace, translate) special characters on input or output. The following mapping flags are supported:

              INLCR   Translate NL to CR on input.

              IGNCR   Ignore carriage return on input.

              ICRNL   Translate carriage return to newline on input (unless IGNCR is set).

              ONLCR   Map NL to CR-NL on output.

              OCRNL   Map CR to NL on output.

              If defining more than one flag, the flags must be comma separated.

The configurations flags are taken directly from the termios man page so there should be little confusion as to what they do.

@larsks Please test this feature and let me know if it works for you - it's on the map_feature branch. If it works for you I will merge it to master. Thanks.

larsks, 2017-11-05:
I appreciate you taking a look at this.

First, it looks like I made a mistake earlier: the devices I'm using (in this case, an esp8266 running micropython) wants just <CR> as the EOL character.

I played a little bit with termios flags before your patch, and despite the description of `ONLCR` it doesn't seem to work as expected.  E.g., given:

    tio -m ONLCR -b 115200 /dev/ttyUSB1

When typing `help<RETURN><RETURN>` I see in `strace`:

```
17608 execve("./src/tio", ["./src/tio", "-m", "ONLCR", "-b", "115200", "/dev/ttyUSB1"], 0x7fffc5023890 /* 70 vars */) = 0
[...]
15869 write(3, "h", 1)                  = 1
15869 write(3, "e", 1)                  = 1
15869 write(3, "l", 1)                  = 1
15869 write(3, "p", 1)                  = 1
15869 write(3, "\n", 1)                 = 1
```

Compared to the same operation in `picocom`:

```
17563 execve("/usr/bin/picocom", ["picocom", "-b", "115200", "/dev/ttyUSB1"], 0x7ffcde103c70 /* 70 vars */) = 0
[...]
17563 write(3, "h", 1)                  = 1
17563 write(3, "e", 1)                  = 1
17563 write(3, "l", 1)                  = 1
17563 write(3, "p", 1)                  = 1
17563 write(3, "\r", 1)                 = 1
```
In `picocom`, hitting `RETURN` executes the line, while in `tio` it has no affect.  I was only to get `tio` to work by adding a new option to explicitly map NL -> CR in `tty.c`; you can see the patch [here](https://github.com/larsks/tio/commit/2473cb8ca4aa87dabb3263bf5ba747ad5dc1c445).  I'm happy to submit that as a pr if you're interested.

lundmar, 2017-11-06:
Thanks for the debug. I didn't expect to see the '\n' there.

It turns out that the default stdin is configured to translate CR -> NL and since tio does not touch the stdin settings we simply inherit this behaviour. However, I want the input to be accurate so I'll make tio reconfigure stdin so that we actually get a CR when hitting enter. This should fix your problem. Then any additional needs people might have should be sufficiently handled by the new --map option.

I'll get back to you with a new patch tomorrow.

lundmar, 2017-11-06:
Ok, I've updated the map_feature branch. Now you should see a `\r` on input :smile:

larsks, 2017-11-06:
That did it! Works like a charm.

lundmar, 2017-11-06:
Great! Thank you for reporting this issue as it helped find and squash this bug.

I'll clean up a few things and prepare a new release of tio soon.

Happy esp8266 hacking :wink:

lundmar, 2017-11-06:
I've published a new release of tio which includes the fix so I'm closing this issue. Thanks for all the inputs.

closed 2017-11-06 by lundmar