tio/tio #66: Simple character translation is needed

janoc, 2017-12-26:
It would be really helpful to have an option to translate received '\n' to '\r\n' for devices that don't send the '\r' character (e.g. many Arduino-based ones). 

Another useful option would be to allow the backspace character to be remapped to actual CTRL+H - e.g. my Buspirate doesn't react to backspace and I have to use CTRL+H to erase character in the built-in shell.

lundmar, 2017-12-28:
I think the first remapping is reasonable because it is required for tio to work with Arduino devices.

However, the CTRL-H remapping seems a bit arbitrary to me and I'm not sure I want to start support all sort of remappings, except the necessary ones. Buspirate fix your backspace 😄

janoc, 2017-12-28:
That is not only a Buspirate problem. Backspace vs DEL is actually a way more common issue when dealing with terminal emulators than the \n vs \r\n problem.

See e.g. here:
http://tldp.org/HOWTO/Keyboard-and-Console-HOWTO-5.html

Most terminal emulators allow to remap this, because messing with stty and what not will usually break other programs (or requires wrapper scripts). 

E.g. Putty allows to remap this, Screen allows to do it, Minicom, etc. Pretty much every single terminal program, because it is so common problem.

Furthermore, it is a fairly trivial fix, all that is needed is a command line option and then a lookup table to use when sending/receiving the characters. That would easily allow support for both the newline remap and this (and any other remap that may be needed in the future - e.g. whether RETURN should send only LF or CR+LF is sometimes required).

lundmar, 2017-12-28:
Hmm. you are of course right. I've just never noticed it myself because I work mostly with Linux targets where this is normally not an issue.

Yes, it is a trivial fix. However, I think I have to consider creating a programmable mapping interface to solve any future remapping needs in one go instead of constantly adding new flags. The current mapping flags are those directly supported and remapped by the tty device itself. Anything else would have to be remapped by tio.

janoc, 2017-12-28:
One solution that I have seen for handling the user interface part of this was for the user to provide a mapping table (= two strings). Kinda like the 'tr' Unix command. 

However, for these very common cases I think also a dedicated option would be good, because having to specify the table only because my Backspace key isn't working right is a pain. Most terminal programs don't attempt to be very smart here - just offer e.g. a checkbox to swap Backspace/DEL or interpret LF as CR+LF, nothing more. 

BTW, Backspace/DEL can be remapped by the tty device itself (there is an stty command for it). There may be one for the CR+LF issue too but I am not too sure about that.

lundmar, 2017-12-28:
I want to keep the tio interface simple so all mappings are specified via the --map option.

For now, what we could do is add support for the following flags:
$ tio --map INLCRNL,OBSDEL ...

INLCRNL: Map NL -> CRNL on input.
OBSDEL: Map BS -> DEL on output.

janoc, 2017-12-28:
OK, that works for me.

lundmar, 2017-12-29:
I've added a branch that features the new mappings here:
https://github.com/tio/tio/tree/new_mappings

Please feel free to test. If it works for you I'll add it to mainline.

In the future I might rework the mapping code to make it fully programmable but this will have to do for now.

janoc, 2017-12-29:
OK, testing the code and there are some buglets:

- when OBSDEL is enabled, it should translate DEL (0x7h) to BS (\b). Right now it does the reverse, so the Buspirate problem is not solved.  The reason for this is that terminal emulator is sending DEL instead of Backspace (good old Unix terminal convention which had a DEL key above Enter, not Backspace). Perhaps you may want to change the name of the option to ODELBS too. I have fixed it here and it works like this: #67

- INLCRNL seems to be working fine.

- The mapping flag names should be probably documented in the --help output - this is something people will need often and the names are not super obvious to remember.

lundmar, 2017-12-29:
He he ok, fixed.

I've put a note to see the man page for mapping flag details - I think that should suffice.

janoc, 2017-12-29:
OK, thanks!

lundmar, 2017-12-29:
Great, I'll merge it in. Thanks for testing! 👍

closed 2017-12-29 by lundmar