tio/tio #69: Baud rate as an optional second positional argument

wavexx, 2018-06-02:
I'd love to be able to specify the baud rate directly as a second positional argument, as screen does:

````
tio /dev/ttyUSB0 9600
````

This would simply be an alternative to using ``-b``.
This would allow me to alias screen=tio for most of the existing arduino tooling as well ;)

arkamar, 2018-06-02:
On Sat, Jun 02, 2018 at 06:49:43AM -0700, wavexx wrote:
> I'd love to be able to specify the baud rate directly as a second
> positional argument, as screen does:
> 
> ````
> tio /dev/ttyUSB0 9600
> ````
> 
> This would simply be an alternative to using ``-b``.
> This would allow me to alias screen=tio for most of the existing
> arduino tooling as well ;)

You can create a simple script accepting arguments as you need, that
transfers them to tio.

I believe, that it is better solution, than introducing baud rate
directly as a second argument.

lundmar, 2018-06-03:
@wavexx , I understand where you are coming from. I used screen like that for many years. In fact one of the main reasons I wrote tio was to avoid running screen under tmux which just seemed silly to me. Also, I wanted something simpler than the alternatives.

That being said, I don't really want to extend the current command line interface with a screen like baud rate option because I think it is quite perfect as it is - having explicitly defined options is actually a good thing. Also, to avoid any ambiguity I don't want two ways of defining the same option.

in your case, I suggest creating a simple bash function that wraps tio to mimic the interface you need or create a script as @arkamar suggested.

wavexx, 2018-06-03:
On Sun, Jun 03 2018, Martin Lund wrote:
> @wavexx , I understand where you are coming from. I used screen like that for many
> years. In fact one of the main reasons I wrote tio was to avoid running screen under
> tmux which just seemed silly to me. Also, I wanted something simpler than the
> alternatives.

I love tio. I'm often using it along with rlwrap, with this patch I've
submitted:

  https://github.com/hanslub42/rlwrap/pull/84

For one, I'm using rlwrap+tio to cook input for live gcode debugging
with a simple cnc machine I've built and 3d printers.

> in your case, I suggest creating a simple bash function that wraps tio to mimic the
> interface you need or create a script as @arkamar suggested.

I don't want to push my view here, I'm fine with the current behavior.

I'm already using a shell wrapper around tio most of the time, but I
have to parse all tio flags in order to forward the arguments properly
of course.

lundmar, 2018-06-03:
I'm glad you like tio and have found a good use for it :+1: 

Though, in this case I have to be the guardian of consistency to make sure tio stays great :wink:

closed 2018-06-04 by lundmar