tio/tio #74: Escape leaves terminal in dirty state

Alexander--, 2018-06-27:
tio version: 1.31

tty into another machine and launch some app, using raw mode (for example, `busybox vi`). Press `Ctrl+t q`. The local terminal window stays filled with scrambled leftovers from `vi`.

lundmar, 2018-06-27:
Scrambled how?

When you quit, tio restores your terminal stdout settings. However, tio does not clear the screen so any leftover from e.g. busybox vi will be preserved.

Alexander--, 2018-06-28:
> However, tio does not clear the screen so any leftover from e.g. busybox vi will be preserved.

@lundmar why not? Can you use an alternate screen buffer to restore terminal contents?

Also is it possible to detect, when remote app tries to enter a raw mode (sorry if this is a stupid question)?

lundmar, 2018-06-28:
It is the fully intended behavior. The way it works tio enters raw mode to support simple I/O. There is no alternate buffer because we want the output of tio to be picked up by the scroll buffer of whatever terminal application users are using. It's simple and very useful/flexible in case you do not log to file and need to go back in your scroll buffer to check what happened.

Also, entering an alternate buffer would imply the use of ncurses and I just don't see that happening - we don't want tio to become like minicom. At this point I don't see the need to enter the alternate buffer dimension :black_large_square:

lundmar, 2018-06-28:
Oh, and AFAIK there is no way to detect when remote app enters raw mode. There are no signaling paths for such communication.

closed 2018-06-30 by lundmar