tio/tio #76: Installing on Mac 10.13.6 Errors

Oteng, 2018-09-03:
i created the configure file using 
`$ sudo autoconf`
but when i run 
`$ sudo ./configure` 
i get this error 
./configure: line 1796: syntax error near unexpected token `1.11'
./configure: line 1796: `AM_INIT_AUTOMAKE(1.11 foreign dist-xz no-dist-gzip -Wall -Werror)'

lundmar, 2018-09-03:
If you do intend to install tio systemwide like that you should only use sudo for the very last install step, that is, sudo make install. However, it is better to install it locally in your home without any sudo use (e.g.: ./configure --prefix=$HOME/opt/tio, make, make install). If available for your distro, you should try out the packaged or snap version.

Oteng, 2018-09-03:
Hi, I have an other issue, after connecting every thing and running 
`$ tio -b 9600 /dev/tty.usbserial`
and get this output and it hangs forever
`[tio 19:43:01] tio v1.31`
` [tio 19:43:01] Press ctrl-t q to quit`

lundmar, 2018-09-03:
Check that you have permissions to access the serial device. Typically you need to be member of the dialout group to access serial devices.