tio/tio #77: Request for option to write console output to a file

kennethburgener, 2018-09-14:
We love tio!  So much easier to use than minicom.

Small feature request - option to write console output to a file.

sly74fr, 2018-09-14:
Hello,
The option is already there !!
Use -l filename 😁
Bye

lundmar, 2018-09-14:
What Sylvain said :+1: 

Use ```--log <filename>``` or shorthand ```-l <filename>``` which will write console output to file. You can also use it in combination with ```--local-echo``` to log input.

Another little trick - I recommend you increase your terminals scroll buffer. tio is written without the use of e.g. ncurses emulation which means that all activity is picked up directly by your terminals (e.g. gnome-terminal) scroll buffer. This oftentimes comes useful in case one forgets to log to file and need to inspect some whatever activity happened earlier.

Oh, I'm glad you love tio too - sometimes less is truly more :smile:

lundmar, 2018-09-18:
@kennethburgener Feel free to close the issue if you got it working as intended ;)

closed 2018-09-19 by lundmar