tio/tio #83: nohup support ?

sly74fr, 2019-03-21:
Dear @lundmar,

Thanks again for tio, I rely on it all day long at work for CERN !
I just noticed today that I can't start it with nonuple as I get a 'Cannot save stdin...' error (sorry for the approximate description, I no longer have under eyes).
Do you think this could be solved ? I went with 'screen' meanwhile...

Best regards,
Sly

lundmar, 2019-03-21:
Hi Sly - I'll look into it. Tio currently requires the standard input to be a terminal but running it under nohup apparently redirects input from /dev/null hence the error.

lundmar, 2022-02-15:
I believe this relates to #87

lundmar, 2022-06-16:
@sly74fr This issue may be fixed by dba77eb9126e0ae1e9b17ee531edd9f4064982d8 . Let me know if it does not work as expected for you.

lundmar, 2022-06-21:
In short, nohup is only suitable for keeping applications in the background which do not process input streams.

closed 2022-06-21 by lundmar

sly74fr, 2022-07-22:
For archiving purpose, the 'faketty' solution was tested successfully with nohup on macOS:
https://stackoverflow.com/questions/1401002/how-to-trick-an-application-into-thinking-its-stdout-is-a-terminal-not-a-pipe/60279429#60279429

AleRoss97, 2025-05-02:
Hi,

First things first: thank you for the awesome software!

Regarding this:
> In short, nohup is only suitable for keeping applications in the background which do not process input streams.

Isn't this the exact use case for logging a serial port? Nohup would  be useful to not manage any session, just logging the output of the serial port.

Best Regards,
Alessandro

lundmar, 2025-05-02:
> Hi,
> 
> First things first: thank you for the awesome software!

I'm glad you like it. Thanks go to the contributors.

> Regarding this:
> 
> > In short, nohup is only suitable for keeping applications in the background which do not process input streams.
> 
> Isn't this the exact use case for logging a serial port? Nohup would be useful to not manage any session, just logging the output of the serial port.

Maybe. It may be worth revisiting issue. I don't remember the details but there was a problem implementing this.

AleRoss97, 2025-05-02:
> I'm glad you like it. Thanks go to the contributors.

Thanks for them too of course.

> Maybe. It may be worth revisiting issue. I don't remember the details but there was a problem implementing this.

Right now i commented this line https://github.com/tio/tio/blob/f887756a7159967e015e3422e9ab04a82db8945f/src/main.c#L76 and added a while(1) pause(1); at this line https://github.com/tio/tio/blob/f887756a7159967e015e3422e9ab04a82db8945f/src/tty.c#L332 to quickly make it work. I'm sure there will be a better, not a hack solution. With that it's working fine for me.

lundmar, 2025-05-02:
If I remember correctly the problem is that we want tio to support commands like this:

```
echo "some data" > tio /dev/ttyUSB0
```
Which will send and exit.

The problem is that the input data stream ends with EOF same as /dev/null input value so we can't distinguish one situation from the other, at least I think the investigation think ended there last time. Further investigation may come up with another solution.

lundmar, 2025-05-02:
I want to avoid an option like that. I think I know how to do it but it will be part of a simplification of the general I/O handling in tio. It will no happen anytime soon but I will put in on the TODO.