tio/tio #92: Local echo doesn't?

larsks, 2019-10-08:
According to the man page (and `tio -h`), the `-e` (`--local-echo`) option should enable local echo...but it doesn't seem to do that. I'm connected to a device that does not echo back characters I type, and regardless of whether I use `-e` or not I see nothing on my screen.

larsks, 2019-10-08:
This is with tio from commit 39a8f63640709aae0b89639a200a6931afd9ab39.

larsks, 2019-10-09:
It looks like this can be resolved by calling `fflush(stdout)` after `putchar(c)` in `print_normal`. I wonder if it would be more effective just to open `stdout` in non-buffered mode?  E.g., by calling `setbuf(stdout, NULL)` in `stdout_configure`.

larsks, 2019-10-17:
There are two pull requests with the two different solutions. I guess the second (#94) is probably the better choice, since I'm not sure if there are negative implications to simply disabling buffering on stdout.

larsks, 2020-06-11:
Given that the last commit is from 2018, my assumption is that nobody is working on this code anymore.

sur5r, 2020-06-11:
The last website change was last year and @lundmar is still active.

I consider applying one of the patches to the Debian package for the time being as I stumbled across this myself.

numericOverflow, 2020-10-12:
I've got the same issue - local echo doesn't work using either the cmd line flag or menu toggle setting

> Linux raspberrypi 4.19.97+
> Distributor ID: Raspbian
> Description:    Raspbian GNU/Linux 10 (buster)
> Release:        10
> Codename:       buster

larsks, 2020-10-12:
@numericOverflow you could try one of the patches I submitted and see if that fixes your problem.

PR #93 (2021-01-18): Disable line buffering in stdout
In order for local echo to work properly, we have to either call
fflush(stdout) after every character or just disable line buffering.
This change uses setbuf(stdout, NULL) to do the latter.

Closes #92
commit 4d4ee466 (2021-01-18): Disable line buffering in stdout

In order for local echo to work properly, we have to either call
fflush(stdout) after every character or just disable line buffering.
This change uses setbuf(stdout, NULL) to do the latter.

Closes #92
PR #94 (2021-01-18): fflush() after putchar() for print_hex and print_normal
In order for local echo to work properly, we have to either call
fflush(stdout) after every character or just disable line buffering.
This change calls fflush() after putchar().

Closes #92
commit c6af6c08 (2021-01-18): fflush() after putchar() for print_hex and print_normal

In order for local echo to work properly, we have to either call
fflush(stdout) after every character or just disable line buffering.
This change calls fflush() after putchar().

Closes #92
closed 2021-01-18 by lundmar