tio/tio #97: Mapping flag for CR to CRLF?

numericOverflow, 2019-12-14:
My serial device sends a CR, but no LF, so anything that I read from the device gets constantly re-written over itself on the same line.  This is obviously not very useful, and the same thing happens in the log file.

I was experimenting with the "--map" argument and I can't seem to find one in the code totranslate a CR output (from device) to CRLF.

The closest is "ICRNL", however it never returns the cursor to column1, so repeated responses from the serial device get pushed farther to the right:
[image]

What I really need is a "ICRCRNL" flag. From what I can tell, this doesn't exist but is there a way to implement it without a custom code change?

sur5r, 2019-12-14:
I think you can combine `ICRNL` and `INLCRNL` as the first one is done by the kernel and the second one by `tio` itself.

numericOverflow, 2019-12-14:
Perfect, that did the trick!

For anyone else who might stumble across this page with the same issue/need, this is the mapping arg that worked for me:
`--map ICRNL,INLCRNL`

[image]

closed 2019-12-14 by numericOverflow

dc740, 2022-08-23:
ah, finally.

Same as the comment above. This fixes missing new lines when connecting to a 3D printer that uses Marlin firmware. 
 Thank you!

abcd-ca, 2023-12-15:
I have a device that expects CRLF after each command but my mac (ssh'd into debian, using tio) was sending just LF. This is the way to map LF to CRLF:
`--map INLCRNL`

Also, you can use `--log tio.log` then in another console do, `watch -n 1 cat -e tio.log` to see the tio log with hidden line ending characters. CRLF will show up as `^M` while only LF shows up as `$`