tio/tio #104: Hexadecimal and timestamps mix with each other

hrw, 2020-05-15:


stappersg, 2020-07-11:
> When hexadecimal mode is turned on in session with timestamps then they mix with each other

and

> Also would be nice to have a way to set width for hexadecimal mode.

I can imagine that original issue reporter is trying to report something like
_when `hexadecimal` is active, should `timestamp` start a newline._

hrw, 2020-07-11:
If timestamp is enabled then each line of output should be:

[tio hh:mm:ss] HEX OUTPUT OF ENTIRE LINE WITH EOL MARKERS PRINTED TOO
[tio hh:mm:ss] HEX OUTPUT OF NEXT LINE WITH EOL MARKERS PRINTED TOO

lundmar, 2022-02-15:
This seems to be fixed. Closing issue.

closed 2022-02-15 by lundmar

Bubu, 2022-09-29:
https://github.com/tio/tio/commit/858b898bbb04fabe8a7fd310310e7ebb7bdb38b0 broke this again. not sure if this was on purpose? (it's basically a full revert of the fix here)

(I think I needed this to be working to do something, but turns out I actually want https://github.com/tio/tio/issues/170)

lundmar, 2022-09-29:
Yes, the hex output is a bit messy at the moment if you enabled both hex timestamp mode.

The hex mode generally needs to be reworked.

In hex mode it does not make sense to add timestamps nor handle newlines. Hex mode should only do simple output of all received bytes and I'm thinking the more traditional hex output format would be more useful. For example:
```
00000000  74 65 73 74 20 74 65 73  74 20 74 65 73 74 20 74  |test test test t|
00000010  65 73 74 64 66 0a 61 0a  66 61 0a 66 0a 61 73 66  |estdf.a.fa.f.asf|
00000020  64 61 64 73 66 61 73 66  64 61 73 64 66 61 64 73  |dadsfasfdasdfads|
00000030  66 0a 61 73 64 66 61 64  73 66 61 73 64 66 61 73  |f.asdfadsfasdfas|
00000040  64 66 0a 61 73 64 66 61  64 73 66 61 73 64 66 61  |df.asdfadsfasdfa|
00000050  73 64 66 66 64 61 73 64  66 0a 0a 31 32 33 31 0a  |sdffdasdf..1231.|
00000060  65 32 31 64 73 77 65 64  0a 0a                    |e21dswed..|
0000006a

```
No timestamps, just byte count in hex, hex bytes, and ascii representation.

lundmar, 2022-09-29:
FYI - I have disabled timestamp in hex mode.