tio/tio #108: Simplicity above color

stappersg, 2020-07-11:
tio has simpilicity in it's design.  (and advertises that simplicity proudly)

How did adding color get in the mix?

Actually:   Why should tio keep color spraying?

Looking forward to your insights.

LeSpocky, 2021-03-08:
Personally, I find it quite nice that messages from tio stand out. Makes it easier to distinguish from the actual terminal input/output.

carpetfuz, 2022-01-23:
Can we can get a --nocolor option?  I'm colorblind and have to adjust colors all the time anyway for a high contrast environment.  Lacking a control here, it takes me a lot of squinting and concentration to see and understand the colored text emitted from tio.  A simple color-free mode would be a big help.  Otherwise IMO tio is great.

carpetfuz, 2022-01-28:
It's fairly easy to alter src/include/tio/print.h to remove color:
```
#ifdef TIO_NOCOLOR
#define ANSI_COLOR_GRAY      ""
#define ANSI_COLOR_RED       ""
#define ANSI_COLOR_GREEN     ""
#define ANSI_COLOR_YELLOW    ""
#define ANSI_COLOR_BLUE      ""
#define ANSI_COLOR_PINK      ""
#define ANSI_COLOR_CYAN      ""
#define ANSI_COLOR_WHITE     ""
#define ANSI_COLOR_RESET     ""
#else
#define ANSI_COLOR_GRAY      "\x1b[1;30m"
#define ANSI_COLOR_RED       "\x1b[1;31m"
#define ANSI_COLOR_GREEN     "\x1b[1;32m"
#define ANSI_COLOR_YELLOW    "\x1b[1;33m"
#define ANSI_COLOR_BLUE      "\x1b[1;34m"
#define ANSI_COLOR_PINK      "\x1b[1;35m"
#define ANSI_COLOR_CYAN      "\x1b[1;36m"
#define ANSI_COLOR_WHITE     "\x1b[1;37m"
#define ANSI_COLOR_RESET     "\x1b[0m"
#endif
```

I tried to split out session I/O to stdout and tio I/O to stderr, but use of the print.h macros within main.c and others isn't universal and some cases leak through.  For the most part, the putc() and printf() of regular session output are fine, but some cases of diagnostic/systemic prints are hardcoded to stdout in other parts of the code (not using the macros in print.h).

The use of these print macros throughout parts of the code (or not) precludes an easy command-line option that is as-compact as seen above.  One approach would be to create state variables holding color values that are used in place of the ANSI_COLOR_xxx macros.  We would also have to change the associated printfs to consume these variables rather than use preprocessor string concatenation as is done now.  Luckily, the use of ANSI_COLOR_xxx macros is very consistent, so finding them is easy and replacing them with variables would almost certainly work.

The last thing I tried was to add a "tionc" binary build target to the makefile, but my automake skills are extremely lacking.  :)  I can certainly whack one in there post-configure, but I don't know how to make a new target show up from the raw autoconf inputs.

So basically, I failed at three-quarters of what I tried, but did get a build of tio without colors.  Great prog, thank you.

stappersg, 2022-01-28:
On Fri, Jan 28, 2022 at 12:00:58AM -0800, William Alexander wrote:
>   ... did get a build of tio without colors.

Consider to apply it your fork.
Consider to make a merge request for it.

Regards
Geert Stappers
The person who opened this issue.
Learnt the "why" for the many colors.
Meanwhile happy with the color information
that indicates 'this text from tio itself, e.g. "connect/disconnect of
USB device'.
In doubt on wether closing this issue.
-- 
Silence is hard to parse

carpetfuz, 2022-01-29:
stappersg: Please see pull request 'add color-free binary build target "tionc" #132'.  This modification adds a new build target "tionc" that runs without any color directives.

stappersg, 2022-01-29:
Having merge request #132 and me understanding the _why for colors_ resulted in closing this issue.

closed 2022-01-29 by stappersg

lundmar, 2022-02-14:
FYI - tio now features a --color option to set the color you want to use.

Also, the default is changed to a bold white color which I believe all color blinds can see.

stappersg, 2022-02-14:
On Mon, Feb 14, 2022 at 01:37:14PM -0800, Martin Lund wrote:
> 
> Also, the default is changed to a bold white color which I believe all color blinds can see.

Assuming that all color blinds use a non-white backgroud.
Assuming that all use a non-white backgroud.

Regards
Geert Stappers
-- 
Every time you assume,
you make an ass of you and me.