tio/tio #110: Would I/O by more than 1 character be accepted?

jepler, 2020-07-29:
Hello.  I'm a developer of CircuitPython and a user of tio.  While it is probably a CircuitPython bug, we've discovered that some of our devices drop characters sent to them when they are sent 1 by 1 (e.g., when pasting or cursoring, so that there are lots of 1-byte usb transactions in a burst, somewhere between .01% and .1% of characters can go missing).

My fellow developers observed that this is terminal software dependent, and we isolated that tio will write() only 1 byte at a time to the device.

I've implemented this in an older version of tio (1.32, packaged in debian stable) and also found that it makes the terminal experience 'feel faster' overall.

Unfortunately, 1.32 is rather old and the patch doesn't apply directly.  Hence, I wanted to ask, is such a change likely to be accepted into tio if PR'd?  Here's the out of date change:
```
--- tio-1.32/src/tty.c	2018-11-03 10:41:52.000000000 -0500
+++ tio+multiwrite/src/tty.c	2020-07-28 19:34:19.111041381 -0500
@@ -43,6 +43,10 @@
 #include "tio/log.h"
 #include "tio/error.h"
 
+#ifndef O_NONBLOCK
+#define O_NONBLOCK (0)
+#endif
+
 #ifdef HAVE_TERMIOS2
 extern int setspeed2(int fd, int baudrate);
 #endif
@@ -189,6 +193,17 @@
 {
     int status;
 
+    int old_flags = fcntl(STDIN_FILENO, F_GETFL);
+    if (old_flags < 0) {
+        error_printf("Retrieving stdin file descriptor flags");
+        exit(EXIT_FAILURE);
+    }
+
+    if (fcntl(STDIN_FILENO, F_SETFL, old_flags | O_NONBLOCK) < 0) {
+        error_printf("Updating stdin file descriptor flags");
+        exit(EXIT_FAILURE);
+    }
+
     /* Save current stdin settings */
     if (tcgetattr(STDIN_FILENO, &stdin_old) < 0)
     {
@@ -538,18 +553,14 @@
 {
     fd_set rdfs;           /* Read file descriptor set */
     int    maxfd;          /* Maximum file descriptor used */
-    char   input_char, output_char;
+    char   input_chars[256], output_chars[2*sizeof(input_chars)];
     static char previous_char = 0;
     static bool first = true;
     int    status;
     time_t next_timestamp = 0;
 
     /* Open tty device */
-#ifdef __APPLE__
     fd = open(option.tty_device, O_RDWR | O_NOCTTY | O_NONBLOCK );
-#else
-    fd = open(option.tty_device, O_RDWR | O_NOCTTY);
-#endif
     if (fd < 0)
     {
         error_printf_silent("Could not open tty device (%s)", strerror(errno));
@@ -627,41 +638,45 @@
         {
             if (FD_ISSET(fd, &rdfs))
             {
+                ssize_t num_read = read(fd, &input_chars, sizeof(input_chars));
                 /* Input from tty device ready */
-                if (read(fd, &input_char, 1) > 0)
+                if (num_read > 0)
                 {
                     /* Update receive statistics */
-                    rx_total++;
[... 153 lines cut ...]
-                    status = write(fd, &output_char, 1);
+                } else {
+                    size_t num_write = write_ptr - output_chars;
+                    status = write(fd, output_chars, num_write);
                     if (status < 0)
                         warning_printf("Could not write to tty device");
+                    if (status != num_write)
+                        warning_printf("Incomplete write to tty device");
                     fsync(fd);
 
                     /* Update transmit statistics */
-                    tx_total++;
-
-                    /* Insert output delay */
-                    delay(option.output_delay);
+                    tx_total += num_write;
                 }
-
-                /* Save previous key */
-                previous_char = input_char;
-
             }
         } else if (status == -1)
         {
```

jepler, 2020-07-29:
https://github.com/adafruit/circuitpython/issues/3152

lundmar, 2022-06-24:
Hi @jepler , I'm sorry for the very late response to your PR. I have been away but now I'm back to maintaining tio and I have been working on bringing tio up to speed and adding new PR'ed features and doing some polishing and refactoring to now finally address your issue.

The latest git now reads input by block and processes the same block for output to minimize I/O overhead. This means if you paste a big block of data in your terminal it will read that block in one go, process it, and send the same block to the serial device in one go, unless --output-delay is used in which case it will fall back to sending byte by byte with a delay in between. Feel free to try it out.

jepler, 2022-06-24:
That's great news, thanks for the update! Feel free to close this issue up, it sounds like it's outdated.

lundmar, 2022-06-24:
Good, closing issue. This improvement will be part of the next official release which I expect will happen after more thorough testing of this change has been performed.

closed 2022-06-24 by lundmar