tio/tio #115: Compilation error with posix-only default shells

Davoodeh, 2020-10-06:
I tried to install the program via AUR and source as well.
I noticed that I run into compilation error with my default shell (`dash -> /usr/bin/sh`).

And I had to temporarily change my default shell to bash again to get it working (`sudo ln -sfT bash /usr/bin/sh`).

the log is below here.
I don't have anything else to say... I just thought I should report this.

Peace.

Example output for a failed compilation is attached: 
<details><pre>
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:6: installing './compile'
configure.ac:4: installing './install-sh'
configure.ac:4: installing './missing'
src/Makefile.am: installing './depcomp'
autoreconf: Leaving directory `.'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking whether B0 is declared... yes
checking whether B50 is declared... yes
checking whether B75 is declared... yes
checking whether B110 is declared... yes
checking whether B134 is declared... yes
checking whether B150 is declared... yes
checking whether B200 is declared... yes
checking whether B300 is declared... yes
checking whether B600 is declared... yes
checking whether B1200 is declared... yes
checking whether B1800 is declared... yes
checking whether B2400 is declared... yes
checking whether B4800 is declared... yes
checking whether B7200 is declared... no
checking whether B9600 is declared... yes
checking whether B14400 is declared... no
checking whether B19200 is declared... yes
checking whether B28800 is declared... no
checking whether B38400 is declared... yes
checking whether B57600 is declared... yes
checking whether B76800 is declared... no
checking whether B115200 is declared... yes
checking whether B230400 is declared... yes
checking whether B460800 is declared... yes
checking whether B500000 is declared... yes
checking whether B576000 is declared... yes
checking whether B921600 is declared... yes
checking whether B1000000 is declared... yes
checking whether B1152000 is declared... yes
checking whether B1500000 is declared... yes
checking whether B2000000 is declared... yes
checking whether B2500000 is declared... yes
checking whether B3000000 is declared... yes
checking whether B3500000 is declared... yes
checking whether B4000000 is declared... yes
checking whether TCGETS2 is declared... yes
./configure: 4426: test: xyes: unexpected operator
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating src/bash-completion/tio
config.status: creating man/Makefile
config.status: creating src/include/config.h
config.status: executing depfiles commands
Making all in src
make[1]: Entering directory '/home/yas/temp/aur/tio-git/src/tio/src'
  CC       tty.o
  CC       options.o
  CC       time.o
  CC       main.o
  CC       log.o
  CC       error.o
  CCLD     tio
/usr/bin/ld: tty.o: in function `tty_connect':
tty.c:(.text+0x1d18): undefined reference to `setspeed2'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:374: tio] Error 1
make[1]: Leaving directory '/home/yas/temp/aur/tio-git/src/tio/src'
make: *** [Makefile:352: all-recursive] Error 1
</pre></details>

sur5r, 2020-10-07:
The problem comes from this line:
https://github.com/tio/tio/blob/39a8f63640709aae0b89639a200a6931afd9ab39/configure.ac#L89

`==` is not POSIX compliant, `=` should be used for string comparison here.

What I don't understand yet is why this problem doesn't show up on Debian when `dash` is configured as `/bin/sh`.

Davoodeh, 2020-10-07:
/bin/sh?
My issue was ALL related to /usr/bin/sh as I mentioned before.
I don't have a saying in any of these but just I guess wildly that maybe in that setup the /usr/bin/sh is bash and only /bin/sh is dash. That's why the problem cannot be reproduced.

sur5r, 2020-10-15:
I think that doesn't matter for Arch as Arch has usrmerge, IIRC. The generated script calls /bin/sh anyway.

closed 2021-01-18 by lundmar