tio/tio #127: Build process is poorly documented

71GA, 2021-04-26:
Here: https://github.com/tio/tio#31-installation-using-release-tarball

     $ ./configure
     $ make
     $ make install

should be fixed to: 

     $ autoreconf -i
     $ ./configure
     $ make
     $ make install

It is sloppy and incomplete to leave it as is... It also does not work as it is currently.

sur5r, 2021-04-26:
What exactly does not work for you? Please post an error message or similar.

lvbuwei, 2021-11-06:
in document,miss
autoreconf -i

sur5r, 2021-11-06:
No, it's not missing. The section is titled "Installation using release tarball". The release tarball already contains a working `configure` script and `Makefile.in`s. There is no need to use `autoreconf`.

When building directly from git, however, it's necessary to run `autoreconf -i` as stated by @71GA. 

The correct way to fix this is to add another section to the readme for building directly from git.

sur5r, 2021-11-06:
Also, this is what `autogen.sh` is all about.

lundmar, 2022-02-15:
The described build steps from source reasonably assumes you are using the officially released source and not the git source. I will provide no description on how to build from git source because if you do decide to use git you should know what you are doing and also notice the autogen.sh script.

Anyway, this is now a mute point because we have just replaced autotools with meson.

closed 2022-02-15 by lundmar