tio/tio #130: Using tee command with Tio...

vboeree, 2021-11-03:
Hi, I was wondering if someone could help me with a problem I’m having when trying to direct output from Tio to my printer. Here is what I enter in Raspbian terminal…
 
tio -b 300 -d 8 -f soft -s 1 -p none /dev/ttyAMA0 | sudo tee /dev/usb/lp0
 
Here is the error message…
 
Error: Saving current stdio settings failed
 
I’ve used the “tee” command with other programs and it worked fine. Any ideas?
 
Thanking you in advance, Vendel Boeree/ K2DSI.

attila-v, 2021-11-04:
hi,
maybe:
`unbuffer -p tio /dev/ttyUSB0 | tee <filename>`
I tried only with file, not with printer, but I think it will be good.
(you have to install `expect` package for `unbuffer`)
regards,
Attila

vboeree, 2021-11-05:
Hi and thanks for the suggestion. I installed "expect" and added "unbuffer -p" to the command. Tio now runs but still doesn't print. I've used the tee command with Minimodem and it worked like a champ. I have a ham radio RTTY terminal unit connected to my Raspberry Pi Zero W via a serial port and it works great but I just can't get it to print on the fly like it did with Minimodem. I also can't use "setterm --foreground green ;" to change the text color...

attila-v, 2021-11-05:
hm, interesting problem
unfortunately I can not reconstruct this issue, because I do not use printer.
but I have some ideas:
- try `script` from `bsdutils` package:`script -e -c 'tio /dev/ttyAMA0' /dev/null |tee /dev/usb/lp0`
- modify tio: change the `stdout` to `stderr` in the `fprintf` functions in the `print.h` include file, then compile a new tio program

Attila

closed 2021-11-19 by vboeree