tio/tio #131: couldn't see timestamp

iysheng, 2021-12-21:
I installed tio with ``sudo dnf install tio``. The version is as below:
``` version
tio v1.32
Copyright (c) 2014-2018 Martin Lund

License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```
When i connect a device with timestamp enable, but i still could not see timestamp?
[image]

sur5r, 2021-12-21:
I'm a bit confused by your message. Your screenshot clearly shows timestamps.

iysheng, 2021-12-21:
> I'm a bit confused by your message. Your screenshot clearly shows timestamps.

oh. I am so sorry，I Couldn't see timestamp    on the pc，now when i look the pic on phone，i can see the timestamp.Maybe it is the theme of the terminal matter.

closed 2021-12-21 by iysheng