tio/tio #145: Insert newline in terminal only?

sdbbs, 2022-04-29:
I am using msys/tio 1.32-1 under MSYS2 on Windows 10, and it works great.

I was just wondering about the following: I am currently talking to a USB serial port via tio, where the serial port only provides output (which `tio` prints on the terminal), and does not handle input.  So, when I have an active `tio` session, when I press keys on the keyboard (including ENTER), nothing happens - as expected.

However, sometimes the logs that I get, get a little confusing to read, and I would like to insert some newlines *only at the terminal* (that, is, they should not be sent via the serial connection), simply to make some visual delimiters in the log that is output via this serial port/session.

Is there possibly some sort of key combo, that would insert a newline only on the terminal, during a `tio` session?

sur5r, 2022-04-29:
You could enable the local echo via `-e` or `--local-echo`. If anything you sent is ignored by the remote side anyway, that would be the easiest way.

sdbbs, 2022-04-29:
Many thanks @sur5r : 

> You could enable the local echo via `-e` or `--local-echo`. If anything you sent is ignored by the remote side anyway, that would be the easiest way.

Great - I was not aware of `--local-echo`, that would indeed be the simplest in my case.

Unfortunately, in MSYS2, I do not get local echo - although, there have always been some problems with terminal stuff in MSYS2. For the record, I've tried also wrapping my call in `winpty`, that is:

```
winpty -- tio --local-echo --baudrate 115200 --databits 8 --flow none --stopbits 1 --parity none /dev/ttyS6
```

... however, this does not help either.

This is, of course, outside of scope of `tio` itself - but if it's OK, I'd like to keep this issue open, in case someone also working in MSYS2 ends up having a solution for this.

Thanks again for the help!

sur5r, 2022-04-29:
I just noticed the version you are using. You need to update to get the fix for #92.

sdbbs, 2022-05-02:
Thanks again @sur5r : 

> I just noticed the version you are using. You need to update to get the fix for #92.

Got it - I had a bit of a challenge compiling `tio` for MSYS2, but eventually got it sorted - notes are in https://stackoverflow.com/questions/72059601/msys2-makepkg-pkgbuild-with-meson-which-directory-is-used-in-package

So, now I have tio v1.37; and again, I can confirm that using call:

```none
tio --local-echo --baudrate 115200 --databits 8 --flow none --stopbits 1 --parity none /dev/ttyS6
```

... in MSYS2 shell (more correctly, the MSYS2 `bash` shell, running in Windows Terminal), again results with a situation, where pressing <kbd>ENTER</kbd> does ***not*** insert a new line in the terminal.

*However*, I discovered by accident, that pressing <kbd>CTRL</kbd>+<kbd>ENTER</kbd> in fact ***does*** insert a newline in the terminal - which is exactly what I wanted!

Nowever, note that you **must** be connected to a serial port - if you just get something like:

```
$ tio --local-echo --baudrate 115200 --databits 8 --flow none --stopbits 1 --parity none /dev/ttyS6
[11:37:05.853] tio v1.37
[11:37:05.854] Press ctrl-t q to quit
```

... without `tio` saying "Connected", then you're not connected to an active serial port session yet - and so neither <kbd>ENTER</kbd> nor <kbd>CTRL</kbd>+<kbd>ENTER</kbd> (nor any other) keypresses, will result with a newline in terminal.

Once `tio` can actually connect in a serial port session:

```none
$ tio --local-echo --baudrate 115200 --databits 8 --flow none --stopbits 1 --parity none /dev/ttyS6
[11:39:46.471] tio v1.37
[11:39:46.472] Press ctrl-t q to quit
[11:39:46.483] Connected
```

... then you can press most keyboard keys (`a`, `b`, etc) and they will be echoed - and of course, in this case, also <kbd>CTRL</kbd>+<kbd>ENTER</kbd> will work (it will insert a new line in the terminal).

So, my problem is solved now - many thanks again for the help!

sur5r, 2022-05-02:
> _However_, I discovered by accident, that pressing CTRL+ENTER in fact _**does**_ insert a newline in the terminal - which is exactly what I wanted!

This seems to be a property/bug/uniqueness of the Windows terminal or tio on Windows. I can not reproduce such behavior on Linux.

closed 2022-05-02 by lundmar