tio/tio #147: Tio will not start without a configuration file.

GWHAYWOOD, 2022-06-04:
As title, if there is no .tiorc or similar file then tio (version 1.38) will not start.

lundmar, 2022-06-04:
This has already been fixed in v1.38 - are you sure you tested with this version?

GWHAYWOOD, 2022-06-04:
Yes:

root@office:~# tio /dev/ttyUSB0
[18:08:33.270] tio v1.38
[18:08:33.271] Press ctrl-t q to quit
[18:08:33.271] Listening on socket /tmp/tio.sock
[18:08:33.272] Saved log to file /var/log/tio/tio.log
[18:08:33.273] Error: Device file is locked by another process
root@office:~# mv .config/tio/tiorc ~/hidden_tiorc
root@office:~# tio /dev/ttyUSB0
Error: unable to parse configuration file (-1)
root@office:~#

lundmar, 2022-06-04:
Hmm, I can't reproduce, not even with the snap version.

Did you compile tio from source? If so please make sure you are on the commit tagged v1.38 and that you do install after compilation in case you missed it.

GWHAYWOOD, 2022-06-04:
Yes, I compiled from source (on a Raspberry Pi 3B+).  I downloaded the .zip archive on May 19th.

Some directory listings, a shell script and the md5sum of the .zip archive:
[code]
root@office:~# l
total 6680
-rw-r--r-- 1 pi   pi   1877743 May 19 14:49 tio-master.zip
drwxr-xr-x 8 root root    4096 May 19 15:47 tio-master
root@office:~# l tio-master
total 64
drwxr-xr-x 3 root root  4096 May 15 21:37 src
-rw-r--r-- 1 root root  3988 May 15 21:37 README.md
-rw-r--r-- 1 root root   130 May 15 21:37 meson_options.txt
-rw-r--r-- 1 root root  1578 May 15 21:37 meson.build
drwxr-xr-x 2 root root  4096 May 15 21:37 man
-rw-r--r-- 1 root root   709 May 15 21:37 LICENSE
drwxr-xr-x 2 root root  4096 May 15 21:37 images
-rw-r--r-- 1 root root 21770 May 15 21:37 ChangeLog
-rw-r--r-- 1 root root  1187 May 15 21:37 AUTHORS
drwxr-xr-x 3 root root  4096 May 19 15:49 subprojects
drwxr-xr-x 8 root root  4096 May 19 15:51 build
root@office:~# ls -lrt /usr/bin/tio*
-rwxr-xr-x 1 root root 27072 Nov  4  2018 /usr/bin/tio-1.32
-rwxr-xr-x 1 root root 54856 May 19 15:51 /usr/bin/tio-1.38
root@office:~# cat /usr/local/bin/tio
#!/bin/bash

/usr/bin/tio-1.38 /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0 -b 9600 -m ODELBS -l/var/log/tio/tio.log
root@office:~# md5sum tio-master.zip 
29922c7baa4c9ebc5de2b8ac6f95cfa3  tio-master.zip
root@office:~# 
[/code]

lundmar, 2022-06-04:
The reason it is not working is because you are basically compiling a pre-release of v1.38.

You need to download the official release tarball from here: https://github.com/tio/tio/releases

GWHAYWOOD, 2022-06-04:
Ugh.  I didn't know I'd done that.  I just clicked on the big green "Code" button on tio's front page on Github and then clicked "Download zip".  Most of the time I try to avoid Github.  I think it would be fine without all the HTML.

lundmar, 2022-06-04:
Ha ha. FYI - what you download there is the snapshot of the git repository of the day you pressed the button, 19th of May.

tio v1.38 was released 3 days ago so there is a lot of fixes you don't have in your zip.

Either way - I'm sure it is fixed so I'm closing this issue.

closed 2022-06-04 by lundmar