tio/tio #148: Command line parsing is a little bit variable.

GWHAYWOOD, 2022-06-04:
Recently I found that

tio /dev/ttyUSB0 -b 9600 -m ODELBS -l/var/log/tio/tio.log

works fine but

tio /dev/ttyUSB0 -b 9600 -m ODELBS -l /var/log/tio/tio.log

(i.e. with a space after '-l')fails because the logfile path is unrecognized.

lundmar, 2022-06-04:
I've looked into the issue. Unfortunately this is a limitation of the getopt_long parsing mechanism that we use for parsing the command-line options. The inconsistency only applies to the options with optional arguments, meaning --log and --timestamp.

The inconsistency only shines through when using short options. For long options it is well documented that you need to do e.g. --log=test.txt

The alternative is to not use optional arguments and instead introduce an additional argument. For example, -l, --log to enable log, --log-filename to specify log filename.

I'm a bit on the fence on this one. On one hand I want to have as few options as possible but on the other I don't like inconsistencies either.

GWHAYWOOD, 2022-06-04:
Well I can live with it if you can. :)

lundmar, 2022-06-04:
Turns out I can live with the inconsistency so I have fixed it in 94764229229760b104b4886ad4ab2e2d7faa5a9a

closed 2022-06-04 by lundmar