tio/tio #149: Tio may delete its own socket.

GWHAYWOOD, 2022-06-04:
If tio has a socket open, another instance of tio (for example started by accident:) deletes the socket.

lundmar, 2022-06-11:
Fixed in ba22191800d42e5ee408e9edd938c853fd7f0e95

closed 2022-06-11 by lundmar

pcc, 2022-08-17:
The behavior as originally implemented was somewhat deliberate. With the new behavior, if tio is terminated abnormally (e.g. crash or system shutdown), the user has to manually delete the socket before tio can start up again.

Maybe we can make this more robust by checking whether the socket is associated with a running instance of tio before deleting it.

lundmar, 2022-08-18:
We definitely can't have that a second instance of tio will destroy an existing running session.

In case tio does crash and leaves a TCP connection hanging it will eventually be released (after TIME_WAIT elapses) and a new tio session can be started. So I think it best to focus on fixing any situation that may lead to tio crashing. I see this is exactly what you have done in #166 👍🏻

pcc, 2022-08-18:
I don't think this is the case for Unix sockets; the actual socket file will stick around if the program is killed.

While we should certainly fix any crashes, I think that tolerating crashes is a part of making software more robust (i.e. the "crash-only software" principle).

What I had in mind for this specific issue was that if the socket exists, we should try to connect to it. If we get `ECONNREFUSED` back, it means that there's no instance of tio on the other end and we can delete it. Otherwise, we print an error.

lundmar, 2022-08-18:
Right, I was a bit fast there and forgot about the unix socket file case. The ECONNREFUSED sounds like a possible way. That or maybe simply put and check for an advisory lock on the socket file like we are doing on the serial device to prevent multiple tio sessions on the same device file. However, I'm not sure if advisory locks works on socket files.

lundmar, 2022-09-11:
@pcc FYI - it turns out advisory locks are not supported in unix socket files so I've implemented a test for stale unix socket file in 8188bb5b7699360c7893c6deaae8a338daeeac1b based on your ECONNREFUSED suggestion. This way we delete any old unix socket file if no one is listening on it.