tio/tio #150: Enhancement suggestion: as an alternative to defining a log file, configure tio to use syslog.

GWHAYWOOD, 2022-06-07:
I feel the need to be able to use syslog facilities - for example this can automatically add timestamps, process information etc., without disturbing tio itself or its users.

If it's a big deal it can be done in other ways of course, it's just more effort.

lundmar, 2022-06-07:
I'm sorry, but that is not a good idea for several reasons.

Firstly, syslog is mainly meant for logging system events from system services such as web server, print server, file server, display server, etc. Tio hardly belongs in that category.

Secondly, most if not all users want to save their tio logs in separate files and not have to extract them from a system log.

Thirdly, I do not want to support two different log facilities, one of which will be rarely used.

Also, some systems do not feature syslog and so tio will offer different features on different systems. I want to avoid introduce that kind of inconsistency for a simple tool like tio.

Lastly, as you pointed out yourself, in the rare case that someone do want to log to syslog, it is literally as simple as:
```
tio /dev/ttyUSB | logger
```

GWHAYWOOD, 2022-06-07:
I can live with that.  At least the thinking is documented now. :)

lundmar, 2022-06-07:
Oh, by the way. By amazing coincidence, the support for piping output from tio to e.g. logger landed just hours before you created this issue ha ha

Piping to tio does not quite work yet but I've put it on the todo list.