tio/tio #156: setspeed.c

zvezdochiot, 2022-07-12:
Hi @lundmar

Maybe add in `setspeed.c`?:
```c
#include <errno.h>
```

lundmar, 2022-07-12:
Right, fixed in dfcda015c3a6bde16a2a2a4716b4a5cb709b215b. 👍🏻

closed 2022-07-12 by lundmar

zvezdochiot, 2022-07-12:
Hi @lundmar .

To automatically close the issue (and leave a link to it), the commit is called `fix #156: Add missing include `.

lundmar, 2022-07-12:
@zvezdochiot Thank you for the hint but I've been using github since the very dawn of github so I'm fully aware that you can close github issues via commit messages 😉. However, in my open source projects I **prefer**/**choose** to keep a clean git history and not clutter it with anything github specific.

zvezdochiot, 2022-07-12:
@lundmar say:
>However, in my open source projects I prefer/choose to keep a clean git history

It's ridiculous to say that in this particular case. Your entire pack file is 95% demo file: https://github.com/tio/tio/blob/master/images/tio-demo.gif :smile: :smile: :smile: 

```sh
git rev-list --objects --all | git cat-file --batch-check='%(objecttype) %(objectname) %(objectsize) %(rest)' | sed -n 's/^blob //p' | sort -r --numeric-sort --key=2 | less
```
```log
2a355c6e6e4540f32f57c2030686bfa299c8d306 3560233 images/tio-demo.gif
3e468bd235a154a8c4c36336c9ee9bbbbe66899d 3520476 images/tio-demo.gif
5106a281e32d6ffa0640272a7bf02e241f4f543b 3220750 images/tio-demo.gif
2cde63ec6ad20c52251d018416b06a2289d92100 1846157 images/tio-demo.gif
dc889dc7f40e0c0af646be287520c6830254197a 1421950 images/tio-demo.gif
6809d2952bb9033b50aabca5de781bfe4c848256 584837 images/tio-demo.gif
bc22fbda71d9f3d9b935d117a188974b9934347f 31182 src/tty.c
43d579625a20abf2c7ff18c3f9feb07766a74ff2 31129 src/tty.c
3ed2ef2ddc8c3241a46a799c9e127154c21da91b 31061 src/tty.c
db7fdfcc7af18deda743b33b0002f07a14c21680 30909 src/tty.c
: 
```

lundmar, 2022-07-12:
Ridiculous? Who are you to judge that!

It is perfectly legal to import binary files in git. In this case I consider the gif animation files part of the project.

The point is, when it comes to commit messages I do not want github syntactical clutter that may or may not stand the test of time. If I migrate to another git hosting service in the future such github formatting become little but useless noise in the git history.

A little advice here, don't try force your way of doing things onto others.

zvezdochiot, 2022-07-12:
Ok @lundmar .

"The trouble is not that a person is mortal, but that he is suddenly mortal.", "The Master and Margarita", Bulgakov M.A.

:+1: :smile: :heavy_check_mark:

lundmar, 2022-07-12:
"An old apple will take you to places you never expected not to visit" - quote Nonsense generator at https://nonsense.x2d.org