tio/tio #157: Slow down gif

horga83, 2022-07-13:
Hi nice application you have.  It would help if you could slow down the GIF so one could have a few more seconds between commands to see the results.

lundmar, 2022-07-13:
Right.

Recently I used gimp trying to get rid of some background graphics in my gif and it pretty much flattened all wait states to 100 ms when I exported it and I didn't notice.

I have restored the original gif which runs slower.

closed 2022-07-13 by lundmar

ondras12345, 2022-07-17:
Many open source projects seem to use [asciinema](https://asciinema.org/) for their demos. It is free and open source, plays like a video (you can pause it, rewind, etc.) and supports copy & paste.
The only disadvantage I see is that it needs JavaScript to play in a web browser.

lundmar, 2022-07-17:
@ondras12345 I actually started out using that but the problem is you can't easily integrated it in github markdown. Meaning you will be transferred to the asciinema site to view it. With gif it just starts playing automatically within the README which I prefer.