tio/tio #163: Quitting tio doesn't work

BaumGuard, 2022-07-19:
Hi there,

I'm using tio to connect to a satellite receiver that can compute commands sent to /dev/ttyUSB0. After having started tio, this is the second line that appears:
`Press ctrl-t q to quit`
When I'm pressing Ctrl+T nothing happens, also not with Ctrl+T+Q or Ctrl+Q.
I don't know if I'm getting the instructions wrong or how can I terminate tio? At the moment I have to open a second terminal and run `sudo killall tio` to stop it.

sur5r, 2022-07-19:
The sequence to quit `tio` is `Ctrl+T Q`, which means press `Ctrl+T`, release `Ctrl+T`, press `Q`, release `Q`. Notice how `Q` is pressed without `Ctrl`.

> At the moment I have to open a second terminal and run `sudo killall tio` to stop it.

Why did you run `tio` as root? `tio` works fine as a normal user. Please check you `udev` rules.

BaumGuard, 2022-07-19:
Alright, that works.

> Why did you run `tio` as root?

Because tio can't connect to the device if I run it without sudo:

```
[23:33:25.940] tio v1.45
[23:33:25.940] Press ctrl-t q to quit
[23:33:25.940] Warning: Could not open tty device (Permission denied)
[23:33:25.940] Waiting for tty device..
```

lundmar, 2022-07-19:
tio can't access your serial device because the access permissions of your serial device is not configured for normal user access.

As sur5r hinted, you need to fix your udev rules so that the Linux device manager creates your serial devices with permissions so that you can access the serial device as normal user. How to do that differs slightly depending on which Linux distribution you are using.

BaumGuard, 2022-07-20:
Thank you for the hint with udev. I have now created a new .rules-file in `/usr/lib/udev/rules.d` with the following line:
`NAME="ttyUSB0" MODE:="0666"`
Now I can connect to the device without sudo.

Is there also a possibility to send just one command to the device and then exit the session automatically? I'm asking because I want to create a Bash script that changes the channel of the receiver at a given time, sending just "Sx" where x is the number of the channel. After that the script should proceed with further commands.

lundmar, 2022-07-20:
Yes, that is quite simple to do. For example:
```
echo "S0" | tio /dev/ttyUSB0
```
That will make tio open the serial device, send the "S0" command, and finally exit.

Depending on your target, it may need special line endings:
```
echo -e "S0\r\n" | tio /dev/ttyUSB0
```

This feature is actually detailed under examples in the tio man page: https://github.com/tio/tio/blob/master/man/tio.1.txt