tio/tio #164: Lower interval of tio releases?

robert-scheck, 2022-07-23:
Is there any chance to lower the interval of tio releases to about one per week? Otherwise, tio releases will only stay in the Fedora and EPEL testing repository and never reach the stable repository, given the lack of testers adding positive karma to them (for having them pushed from testing to stable before staying one week in testing).

sodface, 2022-07-24:
Just to add a data point, I maintain the tio package for Alpine and the release frequency hasn't presented a problem there, yet. Though I'm still relatively new as a package maintainer there, my sense is that the volume of merge requests (which include package version upgrades) is high enough that the queue needs daily grooming or it would quickly get backlogged. And since the volunteer devs are limited in number and have to approve and merge every MR, the time it takes to get a MR merged varies. Right now Alpine is lucky to have Alice over there staying on top of things. I'm not sure she sleeps.

I thought I would pass along her comment from a recent tio MR I submitted, where I noted "The releases keep on coming!"

> i'm glad it sees active development ^^ this is the only serial client i've used that was actually usable for me..

lundmar, 2022-07-24:
I don't mind making less releases if this is considered a general issue by tio package maintainers. I appreciate the work that tio package maintainers are doing and if that will help them, we can make it so.

That being said, I never really expected tio package maintainers for the various distributions to keep up with the fast paced release scheme cycle that we have been hitting. As we know, many distributions have internal processes that can take weeks to propagate package updates. I simply expect tio package maintainers to pick up a new tio version whenever they deem it prudent for them to do so.

This is kind of where distribution models collide. The main reason I'm pushing frequent releases is because it is a fast way to push bug fixes and new features to users of the tio snap and tarball packages which have no propagation delay. This also helps provide quick feedback since tio is a small project and our end users are also our only testing base.

lundmar, 2022-07-24:
> I thought I would pass along her comment from a recent tio MR I submitted, where I noted "The releases keep on coming!"
> 
> > i'm glad it sees active development ^^ this is the only serial client i've used that was actually usable for me..

@sodface Glad to hear that. Let Alice know we appreciate her kind words. It is the only serial client usable for us too 😁

robert-scheck, 2022-07-24:
For Fedora/EPEL, the non-critical path package update process works like this:

 1. Bump version of package in source code management
 2. Trigger package build in sealed environment
 3. Submit resulting packages to testing repository
 4. Package is being pushed to testing repository (takes up to 48 hours)
 5. Wait for trigger that submits packages to stable repository after
     - at least one positive rating (“positive karma”), or
     - package stood in testing repository for 7 days without negative rating (“negative karma”)
 6. Package is being pushed to stable repository (takes up to 48 hours)
 7. Package reached the stable repository (users receive the package via `dnf update`, mirror synchronisation taking place etc.)

If step 3 is performed with a new package while the previous package is still in step 4 or 5, the mentioned previous package gets obsoleted and replaced by the mentioned new package, while neither karma nor waiting time are inherited. If this happens over and over again, the package in the stable repository will become an old release, because no new package gets pushed to the stable repository.

Negative karma is used to flag an update in the testing repository, which prevents the package from being pushed to the stable repository. Each Fedora/EPEL package gets also an update advisory/errata to provide users details about the changes before updating (shown e.g. in graphical updaters). In difference to that, I would summarize the Alpine packaging more as “build and forget”, because there do not seem to be such (or similar) quality assurance steps. Please don't hesitate to correct me if my understanding about Alpine is incomplete or wrong.

Given there has been exactly one (regular) positive karma for a tio package update in the last nearly 5 years, there are either no people rating the package, or there are simply no users at Fedora/EPEL. And if latter applies, I maybe should consider removing the package (because no other package maintainer showed interest in becoming a co-maintainer), which would save my time.

When looking to https://repology.org/project/tio/badges, multiple downstreams already seem to have a hard time keeping the package up-to-date. And even Alpine seems to update only “edge“, so their development tree, but not stable release branches. Because as of writing, Alpine 3.16 is “stuck” at v1.37. In difference to that, I'm trying to provide the latest package to all users for all currently active Fedora/EPEL releases.

In the end, this is just a kind request from myself as a downstream package maintainer of one Linux distribution (well…two, if you differ between Fedora and CentOS/RHEL/Rocky), which you as upstream can follow – or not.

sodface, 2022-07-24:
> Please don't hesitate to correct me if my understanding about Alpine is incomplete or wrong.

You aren't wrong, QA is left up to the package maintainer (and automate tests if available).

> Alpine seems to update only “edge“, so their development tree, but not stable release branches. Because as of writing, Alpine 3.16 is “stuck” at v1.37.

Right, I should have made that distinction. Alpine's policy is a little squishy but generally only security or major bug fixes get back ported to stable branches.

lundmar, 2022-07-24:
Yes, I understand that the package update process is quite involved for some distributions, in particular for Fedora/EPEL.

What I'm suggesting is that you, as package maintainer for Fedora/EPEL, can simply choose to not submit a new package update while a package is already in the update process. In fact, to lessen your own burden, you could simply choose to check for new tio releases only every e.g. 3 months or more and then decide to submit a new package update.

The way I see it, software release cycles and distro package release cycles do not have to happen at the same frequency. It is fully up to the individual package maintainers to decide their release cycle.

> Given there has been exactly one (regular) positive karma for a tio package update in the last nearly 5 years, there are either no people rating the package, or there are simply no users at [...]

I don't think it reasonable to use the Fedora karma upvotes as any indication of how many users there are of a particular package. I mean, take a look at for example minicom, which is probably the oldest and most popular serial client used on Linux. I'm sure it must have thousands of Fedora users yet it has only 9 karma upvotes over 5 years. Compared, tio, a relatively new serial client, has 7 karma upvotes over 4 years. Clearly the karma numbers are pretty inconclusive to how many actual users there are.

That being said, we do have a few factual usage statistics data points for tio from the distribution systems that provides usage data:
 * Snap: 800+ active installations
 * Homebrew: 900+ installations within the last 3 months
 * Github:
   * 953 clones over the last 14 days
   * 6550 visits over the last 14 days
   * tio-1.47.tar.xz - 67 downloads
   * tio-1.46.tar.xz - 154 downloads
   * tio-1.45.tar.xz - 301 downloads

That is a fair bit of activity for this type of application and the numbers are steadily increasing. I don't think I would be wrong to assume Fedora to have similar but proportional amount of users depending on Fedoras share of the Linux distro scene.

Again, I appreciate all the work that package maintainers do because I still believe that the distro native package distribution method is the preferred way for end users.

However, I'm not too keen on putting restrictions on when to do releases because I work in a very agile way for the reasons stated. What I can promise is that there will be no releases within the coming weeks, as I have exhausted my tio TODO list and I'm turning my attention to other projects. That is, of course, unless a critical bug is reported.

closed 2022-07-25 by lundmar