tio/tio #165: [Enchancement Idea] "Single Command Option"

jd-salazar, 2022-08-11:
I've just discovered tio and it is amazing software. I have a very humble idea that I'd like to express: could there be an option for tio to do the following:

- receive a regular connection string, like` tio -e /dev/ttyACM0` for example
- but with some special option, maybe we can call it `-c` for `--command`, then perhaps tio could do the following:
- example: `tio -e -c /dev/ttyACM0 '*IDN?'` when connected to something like this (https://sigrok.org/wiki/Korad_KAxxxxP_series) would do the following:

- Make a connection with `/dev/ttyACM0`
- Send the string "`*IDN?`"
- If there is a result, then place that result on `stdout` (so in the example with a Korad power supply, it would put "KORAD KD3305P V4.2 SN:32477045" on `stdout`

lundmar, 2022-08-11:
Thanks. I see your use case quite clearly. You want to send SCPI commands to test instruments and parse responses.

Actually, you can almost do that with tio today. To send a command to your instrument simply do as mentioned in the man page:

```
$ echo "*IDN?" | tio /dev/ttyACM0
```

This will send the command to the instrument when the serial device is available. However, tio currently does not explicitly wait for a response so you may or may not see a response.

We could, as you suggest, add an option to enter a command mode which will make tio wait for a string response and maybe with an additional timeout option. Maybe also add an option to not display any tio text in the output so you only see the response text.

FYI - I've seen these cheap Korad PSUs before - you get a lot of bang for the buck. However, if you have any LXI/LAN compatible instruments you may also want to take a look a my other open source project: https://lxi-tools.github.io ;)

jd-salazar, 2022-08-11:
Turns out my [minor] issue was that the repository for Raspberry Pi OS is quite out of date. I did see in your examples that 
`echo "*IDN?" | tio /dev/ttyACM0`
is trivial to do, however the apt repo for Raspberry Pi OS results in:
```
pi@raspberrypi2:~/tio/tio-1.47/build $ tio --v
tio v1.32
Copyright (c) 2014-2018 Martin Lund
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```
I'm fairly sure that once I get a compiled tio running, I should be able to pipe properly and get the functionality I need up and going.

So while maybe the idea isn't bad, it might add bloat to the project*; that is absolutely not for me to even weigh in on. I do really appreciate the response. I'll definitely check out lxi-tools! I once wrote an application using PyVISA, but looking back I wish I had known how much of a dependency NI-VISA is--so getting to ditch that would be great the next time I write an application for something like a scope or psu.

*Aside: The main thing I'm seeing with a command option is that in a scripting language you could so something like 

```
import os #should be using subprocess but bear with me
import sys

def do_the_thing(port=sys.argv[1], arg=sys.argv[2]):
    #can always change to have a return value/return stdout if desired
    os.system(f'tio -c {port} "{arg}"')
def main():
    #do tons of things here
    do_the_thing(...)
```

But if that's way too "pansy", then it's probably a waste of resources to implement the feature.

lundmar, 2022-08-11:
No, I don't think it is a bad idea. Yes, tio started out trying to be a super simple and lean serial port tool with a straightforward command-line interface. It still does. However, tio's scope have changed slightly to also focus on the needs of embedded developers and hackers so a feature like this, being able to send command and receive string response on the command line, is a useful feature.

I'll make a quick note here on how it may work. I'm calling it "response" instead of "command" to generalize if for all use cases (some users may not consider what they send a "command").

Example:
```
$ echo "*IDN?" | tio /dev/ttyACM0 --response --response-timeout 100
```
Options:
```
-r, --response                Wait for response line then quit
    --response-timeout <ms>   Response timeout (default: 100 ms)
```

If you have experienced installing gigabytes of VISA or manufacturer instrument software then I think you are going to love lxi-tools - it is a super lean and performance optimized open source alternative to the well known bloat ;)

sodface, 2022-08-11:
@jd-salazar 

> repository for Raspberry Pi OS is quite out of date.

Excuse me for being rude and butting in off-topic, but as the Alpine maintainer of the tio package, I just wanted to point out that Alpine is a great option for Raspberry Pi's IMHO, see download page for Rpi builds:
https://alpinelinux.org/downloads/

Latest stable 3.16.2 has tio 1.37 in the community repo and 1.47 in edge.

lundmar, 2022-08-12:
@jd-salazar I've implemented the line response feature on the following branch https://github.com/tio/tio/tree/response-feature that you are free to test.

In short, it should make it possible for you to do the following:
```
$ echo "*IDN?" | tio /dev/ttyACM0 --response-wait
KORAD KD3305P V4.2 SN:32477045
```
This will make tio send the ID query string to the serial device and print the first received line response and then quit. If no response is received within the default response timeout (100 ms), tio will quit. The default response timeout can be changed via the --response-timeout option.

It is assumed that the serial device will not echo any received input.

The details are elaborated on in the man page and git commit message.

jd-salazar, 2022-08-15:
@lundmar,

Thanks so much. I'll go ahead and test this now. I pulled the branch as so: `git clone -b response-feature --single-branch https://github.com/tio
/tio`

jd-salazar, 2022-08-15:
@lundmar ,
The feature appears to Work on My Machine(TM) when compiled from source:

```
pi@raspberrypi2:~/tio_beta/tio $ echo "*IDN?" | tio /dev/ttyACM0 --response-wait
KORAD KD3305P V4.2 SN:32477045
```

The option, `response`, also appears as it should:
```
pi@raspberrypi2:~/tio_beta/tio $ tio
Usage: tio [<options>] <tty-device|sub-config>

Connect to tty device directly or via sub-configuration.

Options:
  -b, --baudrate <bps>                   Baud rate (default: 115200)
  -d, --databits 5|6|7|8                 Data bits (default: 8)
  -f, --flow hard|soft|none              Flow control (default: none)
  -s, --stopbits 1|2                     Stop bits (default: 1)
  -p, --parity odd|even|none|mark|space  Parity (default: none)
  -o, --output-delay <ms>                Output character delay (default: 0)
  -O, --output-line-delay <ms>           Output line delay (default: 0)
      --line-pulse-duration <duration>   Set line pulse duration
  -n, --no-autoconnect                   Disable automatic connect
  -e, --local-echo                       Enable local echo
  -t, --timestamp                        Enable line timestamp
      --timestamp-format <format>        Set timestamp format (default: 24hour)
  -L, --list-devices                     List available serial devices
  -l, --log                              Enable log to file
      --log-file <filename>              Set log filename
      --log-strip                        Strip control characters and escape sequences
  -m, --map <flags>                      Map characters
  -c, --color 0..255|bold|none|list      Colorize tio text (default: bold)
  -S, --socket <socket>                  Redirect I/O to file or network socket
  -x, --hexadecimal                      Enable hexadecimal mode
  -r, --response-wait                    Wait for line response then quit
      --response-timeout <ms>            Response timeout (default: 100)
  -v, --version                          Display version
  -h, --help                             Display help

Options and sub-configurations may be set via configuration file.

See the man page for more details
```
@sodface you weren't being rude at all, I definitely want to run something like a pi as lean as possible. 

lastly, @lundmar thanks for telling me about lxi-tools. My team is about to get an oscilloscope that is very new, and I'd very much like to test lxi-tools on it.

lundmar, 2022-08-15:
@jd-salazar Great. Thanks for testing and also suggesting the feature. I'm sure this feature will be useful for the use cases where a simple line request/response is needed, such as sending SCPI commands and receive responses.

I'll clean up and merge the feature branch. I expect the feature will be part of tio v2.0 which I am planning to release soon.

Regarding lxi-tools. Maybe your instrument has already been tested to work with lxi-tools - there is a list of tested instruments in the README. lxi-tools makes it easy to send SCPI commands to network attached instruments. It also features a simple but very powerful lua scripting feature where you can create beautiful simple lua scripts to manage your test automation. I prefer lua over python because python is slow and often becomes a mess.

lundmar, 2022-08-15:
Feature merge in e837fd0303ecc925741f9f0eabf8bc58e7fcd492. Closing issue.

closed 2022-08-15 by lundmar

lundmar, 2024-04-13:
FYI - the tio options ```--response-wait``` and ```--response-timeout``` have been removed in latest code base.

Instead, tio now supports Lua scripting and the same effect can be made via scripting.

For example, pipe command to serial device and wait for line response within 1 second:
```
$ echo "*IDN?" | tio /dev/ttyACM0 --script "expect('\r\n', 1000)" --mute
KORAD KD3305P V4.2 SN:32475045
```

This is a much more flexible solution as we can now wait for any type of response.