tio/tio #168: Windows Shows No Devices

DJDevon3, 2022-09-09:
Windows 10

```py
User@PC MINGW64 ~
$ tio --list-devices

User@PC MINGW64 ~
$ tio -L

User@PC MINGW64 ~
$ tio --version
tio v1.47
Copyright (c) 2014-2022 Martin Lund

License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

lundmar, 2022-09-09:
Yes, unfortunately that feature is silent on Windows/MSYS2 because the underlying system does not provide serial devices by ID.

DJDevon3, 2022-09-09:
Yes please. That would make it easier.

sly74fr, 2022-09-09:
Hi,
I think you can extend the Darwin use-case to do so...
https://github.com/tio/tio/commit/2d1e0f1233949dcbc057af40d4e9ab2e2c11db4d
Cheers,
Sly

lundmar, 2022-09-10:
@sly74fr Yup, I've extended the use case to include MSYS2/CYGWIN in 68fc1596542b13980a4ce4f1154683189cd2b463 . That should do the trick.

lundmar, 2022-09-11:
I've tested tio v2.0 in msys2 and the -L feature now works so I'm closing this issue.

closed 2022-09-11 by lundmar