tio/tio #170: Input in hexmode, output in ascii

Bubu, 2022-09-29:
I'm playing around with OBIS protocol devices and they require a request string which contains characters that are easier to send in Hex mode (i.e. CR;LF or \x06) but the output is actually ascii.

Do you think it might make sense to add seperate input/output modes, or is that a bit too much of a special case?

(for now I recompiles a special version of tio that just always does print_normal, even in hex mode)

Bubu, 2022-09-29:
@stappersg I'm not exactly sure what you are trying to tell me, but I don't think this applies here anyway. I'm not aware of how to easily compose separate tools for interactive use.

lundmar, 2022-09-29:
Can you give an example of such a request string?

I understand that CR (0x0D) and NL (0xA) may be easier to input in hex mode but how about the rest of the request string? Are you going to switch between normal and hex input mode to complete a full request string?

I don't mind having a hex input mode activated by a command key if there is a fair use case for it.

Bubu, 2022-09-29:
I'm new to talking to these devices but what I found is that you send `/?!<CR><LF>` (0x2F3F210D0A) followed by `<ACK>000<CR><LF>` (063030300D0A) and then get a reply such as this (This particular device does seem to only need the first one of those :shrug:):

```
T/ISk5MT174-0001
1-0:0.0.0*255(YYYYYYYYY)
1-0:0.0.1*255(XYYYYYYYY)
1-0:0.2.0*255(1.03)
0-0:C.1.6*255(FDF5)
1-0:1.7.0*255(0.010*kW)
1-0:1.8.0*255(XXX*kWh)
1-0:1.8.1*255(XXX*kWh)
1-0:1.8.2*255(0000000.000*kWh)
0-0:F.F.0*255(0000000)
```

Some References, though German unfortunately: https://shop.weidmann-elektronik.de/media/files_public/9d73b590bf0752a5beff32d229d4497d/HowToRaspberryPi.pdf and the readme.pdf in here: https://shop.weidmann-elektronik.de/index.php?page=product&info=24&dl_media=164 via https://shop.weidmann-elektronik.de/index.php?page=product&info=24

stappersg, 2022-09-29:
From the _How To Raspberry Pi .pdf_

Beispiel für fast alle Stromzähler mit der Baudrate 300:
```text
echo -n -e '\x2F\x3F\x21\x0D\x0A' > /dev/ttyUSB0
```
nach max. 2 Sekunden den folgenden Befehl senden:
```text
echo -n -e '\x06\x30\x30\x30\x0D\x0A' > /dev/ttyUSB
```

The `\x2F\x3F\x21\x0D\x0A` in keystrokes: `/`, `?`, `!`, `return-key`,  `control-J`.

The `\x06\x30\x30\x30\x0D\x0A` in keystrokes `control-F`, `0`, `0`, `0`, `return-key`, `control-J`.

(I'm not sure that `return-key` is a `control-M` )

I hope that `tio` does **not** get bloated with a _"hexmode input"_ feature.  (Yeah, I should have expressed that better in first reply to this issue.)

lundmar, 2022-09-29:
@stappersg There is more to it. You can't just go echo stuff to your serial device without configuring the serial port. Meaning you first need to go figure out how to setup your serial port using e.g. the stty util. Or even better, you can skip the complexities of stty and simply use tio like so:
```
echo -n -e '\x2F\x3F\x21\x0D\x0A' > tio -b 300 /dev/ttyUSB0
```
However, the problem is, in both cases, you will not see the response. Using tio you can use the --response-wait option but it will only give you the first line of the response because it will exit after first line ending is received or timeout. This could be solved if we add a response option strictly based on timeout and ignores line endings.

Regarding how to easily input a series of characters in hex mode. There is already a general feature in most applications that can help make it easier to input in hex (or unicode). For example, in most applications you can press ```<ctrl> <shift> u``` followed by your 2 digit hex value then enter. I think you will find that pretty easy to use.

I'm not against adding a hex input mode and I don't think it is bloat, especially since we already have the input mechanism implemented in the existing hex mode and I can see how this feature is useful if you are hacking/testing/attacking a system like this using arbitrary input vectors/data and want to do a little probing by hand. Also, interface wise, it won't justify a new option but we could add it as an in session key command.

stappersg, 2022-10-03:
I would like to know how useful the advice on _control-F_ for `\x06` was.  (I don't have devices as _issue reporter_ has to test it myself.)

closed 2024-04-10 by lundmar