tio/tio #171: 0 bytes sent/received

bradydean, 2022-10-15:
This is probably a user error, but I have a usb-to-rs485 adapter and I'm writing a program to send out commands to a ptz camera. This adapter has leds for tx/rs and when my program is running the tx light is on, but when tio is running alongside it always shows 0 bytes sent/received. Is this expected or am I doing something wrong?

lundmar, 2022-10-15:
@bradydean Hmm, are you opening tio on a device which is already being used by another application?

bradydean, 2022-10-15:
> the serial session created by tio

How do I know which one tio created?

lundmar, 2022-10-15:
You can't use tio to monitor the I/O of a serial device opened by another application. It does not work that way.

bradydean, 2022-10-15:
Interesting, so what is tio used for then? Sending data to a port?

closed 2022-10-15 by bradydean

lundmar, 2022-10-16:
tio is a serial device tool for connecting to and communicating with serial ports such as the one connected to your embedded device (raspberry pi, arduino, etc.). Once connected you can use the TX/RX statistics command to monitor the TX/RX statistics of tios connection but not that of another application.

However, depending on the serial driver and system, you maybe be able to retrieve global TX/RX statistics for a given serial port.