tio/tio #172: No bash completion?

Zorbatron, 2022-10-16:
OS: Ubuntu Server 22.04, Kernel: 5.15.0-50-generic

When I downloaded Tio from APT, bash auto-completion worked fine, but the version in the APT repository is only v1.32, which is quite outdated and didn't work with any sub-configs.

Tio does work, but there seems to be no auto-complete functionality for both compiling from source and downloading off Snap. The '-' options aren't completed and neither are sub-configurations.

lundmar, 2022-10-16:
@Zorbatron The upcoming Ubuntu 22.10 will feature a more recent tio version.

That being said, if you compile tio from source you will have to make sure the bash completion script is installed the right place to be automatically source by the system. Otherwise you can source the bash completion script manually from e.g. your .bashrc file.

I'm testing the tio snap on an Ubuntu 22.10 beta system and auto-completion works out of the box there.

Zorbatron, 2022-10-16:
Figured out that with `pkg-config`, it would be able to discover bash-completion, so that's all set. However, sub-configurations I've set under `~/.tiorc` aren't auto-completed. Did I miss something, or is that not supposed to be a feature?

lundmar, 2022-10-16:
Auto completion of sub-configuration names is not supported.

closed 2022-10-16 by Zorbatron

Zorbatron, 2022-10-16:
Would such a feature be implemented in the future? The reason I'm using tio now, is that minicom acts weird with tmux, e.g changing the pane size. Minicom allows you to set custom configs that would also be auto-completable.

lundmar, 2022-10-16:
Yes, I'm actually implementing it as we speak :)

lundmar, 2022-10-16:
Implemented in a97b2c00cd3f92925d28d1d444910145ad3ed540 . Give it a try. Does not work with sub-configuration names with white spaces but I may fix that in the future.

Zorbatron, 2022-10-16:
Seems not to work, `tio -v` gives 2.2.

Here's my ~/.tiorc file:

```
# Defaults
baudrate = 9600
databits = 8
parity = none
stopbits = 1
color = 10

[switch1]
tty = /dev/ttyS0
baudrate = 9600

[switch2]
tty = /dev/ttyUSB0
baudrate = 9600

[router1]
tty = /dev/ttyACM0
baudrate = 9600
```

Running `tio switch1` does use the correct settings, but pressing tab with `tio sw` doesn't finish it to at least `tio switch` and promt for `switch1` or `switch2`

lundmar, 2022-10-16:
Are you sure your completion script is updated? Make sure you build and install and then source the script.

Zorbatron, 2022-10-16:
Whoops. After re-logging in, it works! Thank you for such a quick modification 😁.

lundmar, 2022-10-16:
No problem - it was a feature that had been on my mind but forgotten again so thank you for reminding me 👍🏻