tio/tio #177: Inline File logging via a key command

cornellcummings, 2022-11-08:
Hello, thanks for your great program. I know you can setup logging before you make a connection. If it possible, logging a current session to a file via a key command I thing it would be very useful (ie Ctl + f to start a log and ctl+x to end it) this will allow you to capture the specific output you are looking for instead of the whole output. of course the log file names can be auto generated.

lundmar, 2022-11-08:
Thank you. Yes, that is a feature we can add.

However I think one command key to toggle logging on/off will suffice. Log file can be specified via --log-file command-line option or log-file configuration word. If not defined it will write to auto generated log filename.

I'll put it on the TODO list.

cornellcummings, 2022-11-09:
Yes having one key would do nicely. Thank you very much for the feedback.

lundmar, 2022-11-23:
You can now toggle the log to file feature using 'ctrl-t f'.

Feature is implemented in 419fbdc3fafc8e47d5ffd8e32130a5d01f523111

Feel free to test.

closed 2022-11-25 by lundmar

cornellcummings, 2022-12-09:
ok thank you ver