tio/tio #178: Tio cannot accept commands such as ctrl-t q if serial write blocks

jepler, 2022-11-18:
Over in CircuitPython, we're trying to solve a problem where a full serial buffer on our device leads to pain interacting with it. Some background:
 * We like Ctrl-C to work similar to standard Python. Our USB stack supports an interrupt character that can schedule a KeyboardInterrupt, but only when our internal receive buffer has space
 * We don't clear or discard our internal buffer except on request (circuitpython API call), because we need things like "pasting code" to work reliably. I'm not sure how this works at the protocol level (I think it's happening at the USB level, not at an emulated RS232 control line level like RTS/CTS), but the net effect is that on linux write() to the serial device will block indefinitely
 * Frequently, 'normally running code' will never actually read characters sent to it

This leads to a problem where, for instance after unintentionally pasting code while already running a program, it becomes difficult to recover the device, since the internal buffer is filled and the in-band CTRL-C character can't be sent by the host computer.

I had proposed adding use of the modem control lines or the "break" out of band signal, which can be sent by tio. However, it doesn't really help because tio will block in write() and never read the command "ctrl-t b". Nor, of course, "ctrl-t q" to quit. So you typically have to open another terminal or process manager to kill it.

The 'send break to recover' change in CircuitPython does actually work, but only so as long as tio has not already blocked on `write()` to the serial device.

Besides somehow fixing tio (to add a variable sized buffer that can accommodate pending data when write would block?), any other ideas/suggestions to make things better for our users in this situation?

lundmar, 2022-11-19:
Hmm, from your description it sounds like tio is blocking indefinitely in the write() call so you can't quit tio. When using tio do you guys enable hard or soft flow control?

I wouldn't expect to see tio's write call block unless flow control is enabled. By default tio does not enable flow control and that should prevent the write call from blocking indefinitely, meaning the serial USB driver would just fire and forget. At least that is the behavior that I see with your typical FTDI chip serial USB devices.

That being said, tio does block processing of input data so if you paste a block of data it will be written to the serial device as a block of data and while writing it will be blocking because tio is a single threaded I/O loop.

Normally, programs which are blocked on I/O can be interrupted using interrupt signals such as the one fired by hitting ctrl-c (SIGINT). However, such signal handling is disabled in tio because the ctrl-c key code must instead be redirected to the remote device which you are connected to.

The only real way to solve this is to split tio's initial input handling into it's own thread and have it react to ctrl-t q so it can forcefully kill itself while the main thread is blocked on the write() call. There will be some queuing of data involved using piping between the two threads.

jepler, 2022-11-19:
@hathach I can't accurately answer the question about flow control with tinyusb cdc, can you?

hathach, 2022-11-23:
@jepler so far the flow control for only DTR and RTS are supported, while  CTS, DSR aren't not supported. So yeah, I think the flow control is not enabled.

lundmar, 2022-11-23:
So, when you see tio becoming unresponsive you are not using tio's --flow option at all? (defaults to none). What is the exact tio command you guys are firing?

jepler, 2022-11-23:
Typical commandline: `tio /dev/ttyACM0`

```
[09:49:23.251] tio v2.4
[09:49:23.252] Press ctrl-t q to quit
[09:49:23.253] Connected

Adafruit CircuitPython 8.0.0-beta.4-28-g12545fb060-dirty on 2022-11-18; Adafruit Feather ESP32-S3 Reverse TFT with ESP32S3
>>> import time
>>> time.sleep(60)
```
At this point, paste a lot of stuff then try to ctrl-t q

Typical `strace tio` output when the USB buffer fills:
```
select(4, [0 3], NULL, NULL, NULL)      = 1 (in [0])
read(0, "z", 1)                         = 1
write(3, "z", 1)                        = 1
fsync(3)                                = -1 EINVAL (Invalid argument)
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=0}, NULL) = 0
select(4, [0 3], NULL, NULL, NULL)      = 1 (in [0])
read(0, "z", 1)                         = 1
write(3, "z", 1
```

At the USB level, the device responds to a URB_BULK URB_SUBMIT request with an URB_BULK URB_COMPLETE response until the device's buffer fills. Then, it stops sending URB_COMPLETE responses. The host continues sending a few URB_SUBMIT messages, but eventually stops. Later, if the data is processed by the device, USB & tio do recover.

wireshark capture of such a session: [usb-buffer-full.zip](https://github.com/tio/tio/files/10077005/usb-buffer-full.zip)

so this is acting in some sense like flow control (it causes write() to block indefinitely) but is happening at the USB layer, not at an RS232 signal layer such as RTS. I think such blocking would also occur e.g., for a more traditional USB-serial adapter to implement pacing of output data at the real underlying baud rate, though in that case it would always "work itself out" after a period of time.

Hum what if you DO paste 1kB of data at 300 baud through a proper USB-serial adapter or real RS232 port, then try to `ctrl-t q`? that should give you ~30 seconds of waiting time.

We'd be happy to put a CircuitPython device into your hands if it'd help you test & understand what's going on.

lundmar, 2022-11-23:
> At the USB level, the device responds to a URB_BULK URB_SUBMIT request with an URB_BULK URB_COMPLETE response until the device's buffer fills. Then, it stops sending URB_COMPLETE responses. The host [...]
Right, normal bulk USB transmission behavior.

> so this is acting in some sense like flow control (it causes write() to block indefinitely) but is happening at the USB layer, not at an RS232 signal layer such as RTS. I think such blocking would [...]

Right, pacing of data is normal for any serial port depending on the baud rate. If you are pasting a lot of data or you are running at low baudrate then tio will become unresponsive until it has finished writing the pasted data. I think this is acceptable behavior but in your case when everything stalls indefinitely we need an exit strategy.

I think I now understand what the confusion is. It boils down to the difference in implementation or topology of the USB serial channel.

If you use e.g. a FTDI USB serial adapter with no flow control the FTDI chip will simply clock out any data written to it at whatever baud rate it is running and never block, regardless of the UART lines being connected to a micro or not.

However, in your case your micro must be running the USB stack which emulates the CDC class serial port so if your micro stalls everything stalls. Not knowing your HW I assumed that you had a frontend USB serial chip which connected to your micro via regular UART lines but clearly that is not the case.

Either way, this stall situation can occur on any chip which runs the USB stack itself so we need tio to always be responsive to the quit key command.

In fact, I think I may be able to do one better and also add a key command feature to tell the serial device to flush any data written but no sent and data received but not read which should take tio out of the write stall state.

I'll try to make the changes necessary asap so you guys can test it.

jepler, 2022-11-23:
I agree it's unusual. I appreciate you working to improve tio so it works better for us. Feel free to tag me when you're ready for changes to be tested.

lundmar, 2022-11-25:
@jepler I've pushed an experimental branch [here](https://github.com/tio/tio/tree/threaded_input) which introduces a separate thread for handing input so that tio should be more responsive to the quit command even when the main thread is blocked in the write call.

The same branch also introduces a new Flush I/O command which may or may not have any influence on your use case.

jepler, 2022-11-30:
This works nicely!  Now I can send the break signal to my board even if the buffers are full. Thank you!

lundmar, 2022-12-02:
It is now on mainline - closing issue.

closed 2022-12-02 by lundmar