tio/tio #179: No way to send/escape prefix key

detly, 2022-11-28:
It used to be possible to send eg. <kbd>ctrl</kbd>+<kbd>t</kbd> to the remote device by pressing <kbd>ctrl</kbd>+<kbd>t</kbd>, <kbd>t</kbd>. In current HEAD (419fbdc3fafc8e47d5ffd8e32130a5d01f523111) it is not. The help text (from <kbd>ctrl</kbd>+<kbd>t</kbd>, <kbd>?</kbd>) now says:

    ctrl-t ctrl-t  Send ctrl-t character

...but this does not work either eg. <kbd>ctrl</kbd>+<kbd>t</kbd>, <kbd>ctrl</kbd>+<kbd>t</kbd>, <kbd>q</kbd> will still quit the local tio session.

This is particularly frustrating if you are using tio to connect to a device, and on *that* device you are also running tio to connect to a peripheral, as I am. I have no way to quit the remote tio session. My current workaround is to use tio out of git instead of the one installed via Snap, or a different client.

Git bisect shows that the commit that stopped this working was 02729c10b197d76223a68bf349aa0c0c4111c6bb:

```none
02729c10b197d76223a68bf349aa0c0c4111c6bb is the first bad commit
commit 02729c10b197d76223a68bf349aa0c0c4111c6bb
Author: Martin Lund <martin.lund@keep-it-simple.com>
Date:   Fri Jul 15 11:58:53 2022 +0200

    Add support for remapping prefix key
    
    Make it possible to remap the prefix key (default: ctrl-t) by setting
    the prefix-ctrl-key variable in the configuration file.
    
    Allowed values are in the range a..z.
    
    Example, to set the prefix key to ctrl-a simply do:
    
    prefix-ctrl-key = a
```

OS: Ubuntu 22.10
tio: from git
cable: Prolific Technology, Inc. PL2303 Serial Port / Mobile Action MA-8910P
config: 115200 8-N-1, no flow control

lundmar, 2022-11-28:
Thank you for reporting the issue. This is a regression, with the new configurable prefix key I didn't account for those who use tio in tio. I'll fix it asap.

lundmar, 2022-11-28:
Fixed in 12f20c84e3f834750e5f5b356d8bfcf1ac2cc135

Feel free to give it a test run.

detly, 2022-11-29:
Works perfectly in 12f20c8, thanks!

detly, 2022-12-05:
You're not going to believe this, but it broke again in the very next commit (well, sort of).

The sequence <kbd>ctrl</kbd>+<kbd>t</kbd>, <kbd>ctrl</kbd>+<kbd>t</kbd>, <kbd>?</kbd> works as expected, bringing up the help menu of the nested tio session (this did *not* happen pre-`12f20c8`). However, <kbd>ctrl</kbd>+<kbd>t</kbd>, <kbd>ctrl</kbd>+<kbd>t</kbd>, <kbd>q</kbd> quits out of the local tio session, not the nested one.

Confirmed with bisect again:

```text
93e6efc00153e625e4ceb8a51e01b324247d312a is the first bad commit
commit 93e6efc00153e625e4ceb8a51e01b324247d312a
Author: Martin Lund <martin.lund@keep-it-simple.com>
Date:   Thu Nov 24 22:52:28 2022 +0100

    Add threaded input handling
    
    To make tio more responsive to quit and I/O flush key command when main I/O
    thread is blocked on output.
```

closed 2022-12-06 by lundmar

detly, 2022-12-07:
Just confirmed this is fixed :+1: