tio/tio #180: Run external program

vpatkov, 2022-12-01:
It would be nice if tio could run external programs, connecting its stdin/stdout to the port. This is mainly for sending and receiving files with sz/rz/etc.

sur5r, 2022-12-01:
`tio` can already create a UNIX socket to let other programs use the serial port. Unfortunately, `rz`/`sz` does not seem to be able to use it...

lundmar, 2022-12-01:
Right, a key command to run an external program. I'm putting it on the TODO list.

vpatkov, 2022-12-03:
> `tio` can already create a UNIX socket to let other programs use the serial port.

Yes, `-S` is a killer feature, but unfortunately you have to use `netcat`/`socat` to interact the socket usually. Unix way is the standard streams. I think it would be a powerful feature if `tio` could on its own connect the standard streams of an external program to the serial port. Then you can, for example, send files as-is with `cat` or in ASCII with `base64`, write simple shell scripts to automatize things (e.g. programming embedded Linux devices via U-boot). `tio` also could accept a command-line option like `--exec <program>` to execute a program/script after connecting to the port, and then exit.

lundmar, 2022-12-03:
FYI, since you mentioned catting of files, right now tio allows connecting via command line streams, so you can do stuff like this:
```
cat <file> | tio /dev/tty_USB0
```

LeeTroy, 2023-05-05:
In Minicom, I normally create an additional file transfer method that `cat` the input file into terminal for our development board anytime without restart Minicom. 280KB firmware image takes about 25 seconds to transfer with B115200.

So I create a `tio` terminal with a unix domain socket, then use `socat` to dump firmware image over it. However, it tooks about 110 seconds to finish the transfer (280KB). Is there any suggestion that I could look into?

Overall the `tio` user experience is very very good and simple!

lundmar, 2023-05-05:
@LeeTroy Please created a new issue so we don't hijack this thread. Thanks.

lundmar, 2024-05-07:
Option to run shell command with I/O redirected to serial device has been added in e75e19eb0054704c953b6480234b54b76d48e107.

closed 2024-05-07 by lundmar