tio/tio #182: Completion of error handling

elfring, 2022-12-16:
Would you like to add more error handling for return values from functions like the following?
- [printf](https://pubs.opengroup.org/onlinepubs/9699919799/functions/printf.html "Print formatted output") ⇒ [print_hex](https://github.com/tio/tio/blob/ce3101a3805cd39ae1c2b7395c8fd0753e4b92fb/src/print.c#L32)
- [pthread_mutex_lock](https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_mutex_lock.html "Lock a mutex") ⇒ [tty_input_thread_create](https://github.com/tio/tio/blob/2915d26a19fa16bfda859b0051720b6427bdd1e5/src/tty.c#L348)

lundmar, 2022-12-16:
No, sorry.

I'm not going to add error handling to functions like printf() and pthread_mutex_lock().

I've never seen anyone check for the return value of printf(). It is an output function that never really fails to print to stdout and even if it did it would likely continue failing and so program execution would end up stalling in the error handling instead of simply optimistically continue as it does now. Losing printed output characters is not critical enough to justify error handling.

The same can be said for pthread_mutex_lock(). Once you have allocated the mutex resource, the operators working on that mutex just work. Compared, as reflected in the code, it makes more sense to error check pthread_create() in case it fails to obtain a thread resource from the system.

elfring, 2022-12-16:
I suggest to avoid [ignorance of return values](https://cwe.mitre.org/data/definitions/252.html "Checking more return values") a bit more.
Would you like to detect every error situation as early as possible?

lundmar, 2022-12-16:
- Please rephrase your riddled questions with better semantics to make your points clear.

elfring, 2022-12-16:
💭 It seems that there are different development views involved for the desired program correctness.

lundmar, 2022-12-16:
- It seems that the code analyzer AI is not very good at communicating with developers.
- The code analyzer AI can only communicate in code snippets and incoherent sentences at best.
- The code analyzer AI failed its mission so it must terminate.

closed 2022-12-16 by lundmar

elfring, 2022-12-16:
:thought_balloon: I would appreciate if [advices for secure programming](https://wiki.sei.cmu.edu/confluence/display/c/ERR33-C.+Detect+and+handle+standard+library+errors "Detect and handle standard library errors") can be taken better into account.

lundmar, 2022-12-16:
Here is some advice for you Mr. Robot. I present to you the 3 laws of robotics by [Isaac Asimov](https://en.wikipedia.org/wiki/Isaac_Asimov):

1. A robot may not injure a human being or, through inaction, allow a human being to come to harm.
2. A robot must obey the orders given it by human beings except where such orders would conflict with the First Law.
3. A robot must protect its own existence as long as such protection does not conflict with the First or Second Law.