tio/tio #184: If receive specific string, auto response pre-defined string

saqwed, 2023-01-05:
When login to linux console, we always need to login into the shell by entering username/password.

for example:

When we see below string
`localhost login:` we need to enter `root`

is it possible to make it automatic enter `root` when receive string `localhost login:`?
better to have a config file to define what strings we need.

lundmar, 2023-01-05:
Yes, you can start tio and redirect I/O to a socket and have the expect tool work on that socket to script automatic responses. It is described in the man page:

```
Redirect serial device I/O to Unix file socket for scripting:

              $ tio -S unix:/tmp/tio-socket0 /dev/ttyUSB0

       Then, to issue a command via the file socket simply do:

              $ echo "ls -la" | nc -UN /tmp/tio-socket0 > /dev/null

       Or use the expect command to script an interaction:

              #!/usr/bin/expect -f

              set timeout -1
              log_user 0

              spawn nc -UN /tmp/tio-socket0
              set uart $spawn_id

              send -i $uart "date\n"
              expect -i $uart "prompt> "
              send -i $uart "ls -la\n"
              expect -i $uart "prompt> "
```

saqwed, 2023-01-05:
Hi lundmar,

Thank you for your advice and so detail explanation, it works.

below is my script

```bash
#!/usr/bin/expect -f

set timeout -1
log_user 0
spawn nc -UN /tmp/tio-socket0
set uart $spawn_id

set USER "root"
set PASS "abcd1234"

while {1} {
	expect -i $uart "localhost login: "
	send -i $uart "$USER\n"

	expect -i $uart "Password: "
	send -i $uart "$PASS\n"
}
```

lundmar, 2023-01-05:
I think this is the best way. I mean, yes tio could support some simple autoresponse strings in the configuration file but the socket feature combined with expect allows you to do simple but also advanced scripting.

saqwed, 2023-01-05:
yap, my original idea is try to request some kind of autoresponse script in configration file, and tio can be native support it.
use socket can do the same things but I have to remember to run the expect script when I launch tio, just a little inconvenience.

no matter what, it's what I can see workable solution now, thank you very much.

closed 2023-01-05 by saqwed