tio/tio #186: `--timestamp` not in manpage

cvvletter, 2023-01-16:
Dear lundmar/tio team, I was checking the `man tio` page to see if the `--timestamp` option can show more detail than only hh:mm:ss (I don't think it can), but found that the `-t` option is not in the man page at all. Is this something I could help fix? I am not familiar with linux man pages. Thanks in advance!

lundmar, 2023-01-16:
Hi @cvvletter , the -t,--timestamp option is only for toggling the timestamp feature. You use the --timestamp-format to pick which predefined timestamp (24hour, 24hour-start, 24hour-delta, iso8601) format you want. It is documented in the man page.

lundmar, 2023-01-16:
Maybe you are reading a very old man page.

cvvletter, 2023-01-16:
My apologies, it seems my Ubuntu 20.04 package manager ships tio v1.32..., which is the last version not including timestamp miliseconds haha.... Thanks for the quick response!

closed 2023-01-16 by cvvletter

lundmar, 2023-01-16:
I highly recommend upgrading. A lot of changes and new features have been added since v1.32 which is now 4+ years old! ;)

cvvletter, 2023-01-16:
I did (via snap, easiest install for me)! Works perfectly, including milliseconds in the timestamp. Thanks for all the hard work!