tio/tio #187: Snap Install Tio - Using Classic Confinement

bhass1, 2023-01-28:
```
dev1@dev1-desktop:~$ snap install tio
error: This revision of snap "tio" was published using classic confinement and
       thus may perform arbitrary system changes outside of the security
       sandbox that snaps are usually confined to, which may put your system at
       risk.

       If you understand and want to proceed repeat the command including
       --classic.
```

Is there any work being done to have the tio snap use "strict" confinement instead of "classic," and then use defined [snap interfaces ](https://snapcraft.io/docs/snap-confinement#interfaces)?

Are there technical limitations?

lundmar, 2023-01-28:
Last time I checked up on the the snapcraft framework it was required to specify exactly which serial device to allow connection to. This is of course unacceptable for a tool like tio which needs access to any serial port.

bhass1, 2023-01-29:
Yeah, agreed that is unacceptable to specify exactly which serial device to allow connection to.

I didn't look too deeply into this yet, but I think it could be considered an enhancement of tio to get rid of this error by adhering to "strict" confinement rules.

Could also be helpful for the time being to have a note about this in the README under "4.2 Installation using snap (Linux)."

lundmar, 2023-01-29:
Pull requests are welcome ;)

bhass1, 2023-01-29:
Yes, of course :)

I'm trying not to make commitments that I won't follow through with, but at the same time provide useful input. I'll see what I can do tomorrow ;)

closed 2023-01-30 by lundmar

KhazAkar, 2024-05-26:
I'd request opening it again, since there's an appropriate plug available right now - `raw-usb`. I have locally experimental build of tio as snap using strict confinement, and it seems to work fine. To get it to snap store live, there's a simple process on the forum to request auto-connection for such snap, since this plug is a system, privileged one.

Docs:
https://snapcraft.io/docs/raw-usb-interface
https://snapcraft.io/docs/system-interfaces

lundmar, 2024-05-27:
tio is not only about connecting USB tty devices. tio must be able to connect to any tty device and as long as the snap framework has no policy solution for that we will have to continue use classic confinement.