tio/tio #195: Serial Modems Non-interactive

Zyplonox, 2023-04-19:
Hi,

thanks for the work, tio is really great :)

Im working with usb modems using also serial connection. We need to check the IMEIs sometimes which works easy interactive:
```
root@de-bw02:~# tio /dev/ttyUSB2
[23:24:57.617] tio v2.5
[23:24:57.618] Press ctrl-t q to quit
[23:24:57.619] Connected
At+CGSN
XXXXXXX9324538

[23:25:03.897] Disconnected
```
Writing the AT command and press enter, then he outputs the IMEI.

However i tried it non-interactive in some ways:
```
root@de-bw02:~# echo "At+CGSN" | tio /dev/ttyUSB2 -r
At+CGSN
       root@de-b
```
```
root@de-bw02:~# echo "At+CGSN" | tio /dev/ttyUSB2
[23:32:29.144] tio v2.5
[23:32:29.144] Non-interactive mode enabled
[23:32:29.144] Press ctrl-c to quit
[23:32:29.146] Connected
[23:32:29.146] Disconnected
```
```
root@de-bw02:~# tio /dev/ttyUSB2 <<< "At+CGSN"
[23:32:34.695] tio v2.5
[23:32:34.695] Non-interactive mode enabled
[23:32:34.695] Press ctrl-c to quit
[23:32:34.696] Connected
[23:32:34.697] Disconnected
```
Should not this just work with the -r option?

Thanks for help.

Cheers
Jan

lundmar, 2023-04-19:
Hi Jan. I'm glad you like tio :)

The problem is that your modem echoes back whatever is sent so tio quits when the modem echoes back the string you just sent.

The solution is to disable echo in your modem. I expect something like this will work:
```
echo "ATE0" | tio /dev/ttyUSB2
sleep 0.5
echo "AT+CGSN" | tio /dev/ttyUSB2 -r
```

And of course, if you have modem echo disable and want to play around in tio's normal/interactive mode, then you can enable tio's local echo (--local-echo) so you can see what you type.

Zyplonox, 2023-04-20:
Hi lundmar,
thanks for the quick response.
Ah yes i was thinking something like this could be the problem, but i was not aware that disabling the echo in the modem is possible, thanks for that.
 
But trying:
```
root@de-bw02:~# tio /dev/ttyUSB5
[09:53:26.929] tio v2.5
[09:53:26.929] Press ctrl-t q to quit
[09:53:26.930] Connected
ATE0
OK

OK

[09:53:41.562] Disconnected
root@de-bw02:~# echo "At+CGSN" | tio /dev/ttyUSB5 -r
root@de-bw02:~# 
```
He just dont output anything anymore ^^

After the first `OK` i made it persistent with `AT&W`.

But when i wait for the response with `--response-timeout` he fully waits for that timeout. Maybe useful, im not sure.
```
root@de-bw02:~# echo "At+CGSN" | tio /dev/ttyUSB5 -r --response-timeout 10000
root@de-bw02:~# 
root@de-bw02:~# echo "At+CGSN" | tio /dev/ttyUSB5 -r --local-echo
At+CGSN
       root@de-bw02:~# 
root@de-bw02:~# 
``` 
Interactive still just working:
```
root@de-bw02:~# tio /dev/ttyUSB5
[10:05:45.688] tio v2.5
[10:05:45.688] Press ctrl-t q to quit
[10:05:45.689] Connected

XXXXXXX9323555

[10:05:50.584] Disconnected
root@de-bw02:~# 
```
Jan

lundmar, 2023-04-20:
Hmm, looks like the device is not responding to the command.

Try this:
```
echo -e "AT+CGSN\r\n" | tio /dev/ttyUSB5 -r --response-timeout 1000
```

Zyplonox, 2023-04-20:
Same, but he does not wait for the timeout anymore.
```
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB5 -r --response-timeout 1000
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB5 -r --response-timeout 10000
root@de-bw02:~# 
```
btw i need to give the `T` in lowercase, elseway he response with `ERROR`.

lundmar, 2023-04-20:
Hmm, it seems to me the command is sent. Then, maybe before the modem prints the serial number it may be sending a carriage return. When waiting for a response tio will print whatever is received until either a carriage return (CR) or new line (NL) character is received. That would explain why tio may quit before the serial number is received.

It may be a mistake design decision to have tio quit on a received carriage return when waiting on a response string. Maybe it should only quit on a newline. I believe that all systems (DOS vs UNIX, etc.) will always send their string responses with either a NL or CRNL line termination, so waiting only for NL should be good and it may solve your problem.

In the tio source, you could try remove the CR part:
https://github.com/tio/tio/blob/b5ca54c56e2f1648b0f9fd86cf8a524580738e10/src/tty.c#L1429

Replace it with:
```
if (input_char == '\n') 
```

Zyplonox, 2023-04-20:
```
root@de-bw02:/opt/tio-source/tio# cat -n src/tty.c | grep 1429
  1429                              if (input_char == '\n')
```
Now i just get a new line:
```
root@de-bw02:/opt/tio-source/tio# meson install -C build
ninja: Entering directory `build'
ninja: no work to do.
Installing src/tio to /usr/local/bin
Installing /opt/tio-source/tio/build/man/tio.1 to /usr/local/share/man/man1
root@de-bw02:/opt/tio-source/tio# tio -v
tio v2.6
root@de-bw02:/opt/tio-source/tio# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB5 -r --response-timeout 10000

root@de-bw02:/opt/tio-source/tio# echo -e "At+CGSN" | tio /dev/ttyUSB5 -r --response-timeout 10000
root@de-bw02:/opt/tio-source/tio# echo "At+CGSN" | tio /dev/ttyUSB5 -r
root@de-bw02:/opt/tio-source/tio# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB5 -r --response-timeout 10000

root@de-bw02:/opt/tio-source/tio#
```

lundmar, 2023-04-20:
That is puzzling.

I beginning to suspect that maybe your target is simply just sending an extra newline before printing the serial number.

In interactive mode, you could try enable hexadecimal mode to see exactly what characters your modem is receiving and sending.

Zyplonox, 2023-04-20:
```
root@de-bw02:/opt/tio-source/tio# tio -x /dev/ttyUSB5
[11:50:54.756] tio v2.6
[11:50:54.757] Press ctrl-t q to quit
[11:50:54.758] Connected
41 74 2b 43 47 53 4e 
[11:50:57.168] Warning: Invalid hex character: '13' (0x0d)

[11:51:07.630] Disconnected
```

Zyplonox, 2023-04-20:
Thank you very much for the help.

```
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB5 -r
At+CGSN
ERROR
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB5 -r
At+CGSN
XXXXX5059323555
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB5 -r
At+CGSN
XXXXX059323555
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB5 -r
At+CGSN
XXXXX059323555
```
Sometimes the first try is an error. (for USB2 echo is disabled)
```
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB5 -r
At+CGSN
XXXXX059323555
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB2 -r

ERROR
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB2 -r

XXXXXXXX9324538
root@de-bw02:~# echo -e "At+CGSN\r\n" | tio /dev/ttyUSB8 -r
At+CGSN
XXXXXX059323639
```

Zyplonox, 2023-04-20:
The modem is a qualcomm but with custom firmware. 
Thanks for the hint, i will have a look into this.

closed 2023-04-20 by Zyplonox