tio/tio #196: Feature Request: Add option to append or overwrite the log file

arabed, 2023-04-26:
Since most of the time we run/re-run the tio, it will be nice to have the following two features:
1) Add an option to append or overwrite the log file 
2) have a feature to generate log files with the given name and an auto generated number added to the file name e.g. log-file01.txt, log-file02.txt, log-file03.txt,... where the --log-file log-file%%.txt (%% are as an example)
Thanks!

lundmar, 2023-04-26:
I'll make it so that tio by default will overwrite any existing log file. Then add a ```--log-file-append``` option which will append instead of overwrite.

I'm not to keen on adding handling of enumerated log files because tio can't easily keep track of the enumerated files when you start and stop tio - this is something better handled by scripts that may start and stop tio in the same script session.

You are aware that if you only use --log option it will automatically name log files uniquely using a timestamp format?

arabed, 2023-04-27:
Cool! append option will be great! Also while using `--log`, do we have the option to set the log files path?

lundmar, 2023-04-27:
In these matters it is best to consult the tio man page 😉 
```
       -l, --log

              Enable log to file.

              If no filename is provided the filename will be automatically generated.

           --log-file <filename>

              Set log filename.

           --log-strip

              Strip control characters and escape sequences from log.

```

lundmar, 2023-04-27:
FYI - I've updated the man page so it is a bit more specific regarding the log options:
```
       -l, --log

              Enable log to file.

              The   filename   will  be  automatically  generated  using  the  following  format  tio_DEVICE_YYYY-MM-
              DDTHH:MM:SS.log.

              The filename can be manually set using the --log-file option.

           --log-file <filename>

              Set log filename.

           --log-strip

              Strip control characters and escape sequences from log.
```

lundmar, 2023-04-28:
Feature added in 16cba00c

closed 2023-04-28 by lundmar