tio/tio #197: Are considering adding 'Installation using flatpak'?

xin5683, 2023-05-31:
I have always felt that snap is not as elegant as flatpak, so I made a request rather presumptuously.

lundmar, 2023-05-31:
You are welcome to create and maintain a flatpak recipe. I can create a tio.flatpak repository that you can access if you wish.

xin5683, 2023-06-01:
Wow, I'd be happy to help, but I need some time to learn. I'll study how to submit app to Flathub on the weekend when I have some free time.

lundmar, 2023-06-01:
That's great - take all the time you need.

FYI - snap is not as bad as people think. It is simply a different design from flatpak etc. which makes it possible for maintainers to distribute software instantly to users because of a more aggressive security model. With flatpak and normal package distribution channels you have to go through a human review process and that takes time. Time which I can't afford to spend across my multiple open source projects.

That being said, I'm not anti flatpak but I need someone to step up and maintain a flatpak offering :)

lundmar, 2023-06-01:
FYI - I've created https://github.com/tio/tio.flatpak and given you maintainer access to the repo.

lundmar, 2023-06-01:
Happy hacking 👍🏻

lundmar, 2023-09-16:
Feel free to reopen the issue if/when you create a flatpak.

closed 2023-09-16 by lundmar