tio/tio #201: Command History

KindDragon, 2023-08-18:
CoolTerm http://www.coolterm.org supports two terminal modes: "Raw Mode" and "Line Mode". Line mode support command history, but enter command in separate field. In line mode it send command only when you press Enter.

lundmar, 2023-08-18:
I've added it to the TODO file.

Unfortunately I'm fully loaded with other projects at the moment so expect it will be a while before I can add this. Unless of course some one else adds it.

KindDragon, 2023-10-24:
BTW [CoolTerm](http://freeware.the-meiers.org) support it in line mode terminal mode

lundmar, 2024-04-16:
FYI - tio now supports independent input and output modes and this includes a line mode.

For now the line mode is very basic as it only supports backspace editing.

lundmar, 2024-06-15:
History feature has been added in 9d36756dfcd660b0d9d8d199d72d8ac892af91ad. Closing issue.

closed 2024-06-15 by lundmar