tio/tio #203: Add optional Ctrl-C/Ctrl-D/~. exit support

byteit101, 2023-09-06:
I use tio for highly embedded systems where there isn't a real shell, and where I'd prefer Ctrl-C and/or Ctrl-D to exit tio. I just had to look up the documentation yet again to figure out how to exit. I would like some sort of flag, maybe `--simple` or such, that captures ctrl-c/d to exit.

lundmar, 2023-09-06:
I understand what you are saying but I think it is a bad idea to start changing tio's default interaction behavior - it will introduce confusion and inconsistency.

The principle of using a prefix key to activate any tio command is a good one for this particular type of application because in most cases you want all your key strokes to be directed to the target that you are connected, especially ctrl-c, ctrl-d.

That being said, yes, using a prefix key is different from your normal ctrl-c, ctrl-d behavior and it takes a little getting used to and the difficulty of the exit situation is a classic one well known from other applications that use prefix keys (vi, screen, tmux, etc.). This is exactly why we print the exit hint every time you start tio, so that new users can quickly familiarize themselves with the concept and be able to exit:

```
lundmar@wopr:~$ tio /dev/ttyUSB0
[06:42:46.561] tio v2.6
[06:42:46.561] Press ctrl-t q to quit
[06:42:46.562] Connected
```

I think learning how to use the prefix key is a fair minimum requirement for using tio and it is a small hurdle that users must overcome.

byteit101, 2023-09-06:
> changing tio's default interaction behavior

To be clear: I'm not suggesting this. This is about adding an optional, non-default flag

> in most cases you want all your key strokes to be directed to the target that you are connected

In some cases yes, but as of late I've been doing lots of development on things where this isn't the case. All the serial utilities make this assumption, which is fine for shells, but neglects other uses cases. I've been searching around for "bidirectional cat" and other such simple serial programs, but I haven't really found anything in that niche. Adding a flag for tio lets tio fill that niche.

> This is exactly why we print the exit hint every time you start tio

As a person who likes good ux, I saw and appreciated it, but recently my terminals have been dumping data out, so it scrolls off the screen, and then 2 weeks later I come back and just press `(x)` on the terminal window

byteit101, 2023-09-06:
> This is exactly why we print the exit hint every time you start tio

Actually, it would be really helpful if it was also printed on disconnection. I unplugged my device in hopes of figuring out how to exit

sskras, 2023-09-06:
I also think that implementing this as an option (activated on the users' demand) wouldn't hurt the default behavior / UX.

tagraf, 2023-09-09:
Original:  "Press ctrl-t q to quit"
Revision: "Press [ctrl+t] then press [q] to quit"
----
Reason:  I've pressed (ctrl+t) too many time and got confused why it doesn't exit tills I read carefully here in this github page and was scratching my head tills I see 'ctrl-t q' that gave me a hint that I might need to press 'q' after I do 'ctrl+t' and manages to exit it.  But it's not clear since I struggled with the misunderstanding of (ctrl+t) to exit.

lundmar, 2023-09-16:
I'm not going to add multiple ways to exit tio. In tio's interactive mode, the only way to send commands to tio is to use the prefix key, this includes the quit command.

That being said, this is mostly a problem for new users getting to know tio, so I have adopted your suggestion of making the quit hint more explicit in 93e49ab5a23b40bd27e60a1028592ac23e4433f1. However, printing the quit hint on every re-connect/disconnect will result it too much spamming so we won't do that.

closed 2023-09-18 by lundmar